robbinjanssen/home-assistant-omnik-inverter #1: Nice custom component

gerard33, 2020-02-18:
Looks good. Thanks for sharing this.
Maybe you can also add the total power as that is available in the js as well?

robbinjanssen, 2020-02-18:
Great idea, added in v0.0.2! 👍

gerard33, 2020-02-18:
Great!
The Omnik inverters can show data in 2 ways:
```
var webData="NLDN**2017******,NL1-V1.0-0118-4,V2.0-0028,omnik4000tl ,4000,584,345,33734,,4,";
var myDeviceArray=new Array(); myDeviceArray[0]="AANN3020,V5.04Build230,V4.13Build253,Omnik3000tl,3000,1313,685,9429,,1,";
```
My inverter shows it in the second way, so with myDeviceArray. You code only can use option 1 isn't it? The relevant data is in the same positions, so should be an easy addition.

robbinjanssen, 2020-02-18:
I can definitely make that work, does your output from status.js shows 'webData' anywhere?

gerard33, 2020-02-18:
There are some `webdata_*` parts in the js, but no exact `webData` match.

This is the top of the js file for me where I have changed some parts with ---

```js
var version= "H4.01.51MW.2.01W1.0.65(2018-02-271-D)";
var m2mMid = "---";
var wlanMac= "---";
var m2mRssi= "24%";
var wanIp= "192.168.1.127";
var nmac= "---";
var fephy= "off";
var yz_device_type="258";
var yz_device_num="1";
var init_inv_type="0";
var myDeviceArray=new Array();
myDeviceArray[0]="NLRN---BG---,V5.02Build237,V4.13Build252,Omnik3000tl ,3000,0,0,1724,,1,";
;
function initPageText()
{

```

robbinjanssen, 2020-02-18:
And there are no new lines in the actual output right?

I think something like this might work, replace line 82 with:
```python
        """Find the webData."""
        if result.find('webData') != -1:
            matches = re.search('webData="(.*),";function', result)
        else
            """Or the myDeviceArray."""
            matches = re.search('myDeviceArray[0]="(.*),";function', result)
```

Might not work as I see that (if new lines are removed), the string is: `...,3000,0,0,1724,,1,";;function initPageText()...`

In that case the last line of the change should be:
`matches = re.search('myDeviceArray[0]="(.*),";;function', result)`

Could you test this? If it works i'll tag a new release :-)

gerard33, 2020-02-18:
I haven't changed any new lines in the output.

Maybe this can help in verifying your code -> https://github.com/sincze/Domoticz-Omnik-Local-Web-Plugin/blob/master/plugin.py#L147.

robbinjanssen, 2020-02-19:
Seems like that could work, i'll look into it!

robbinjanssen, 2020-02-24:
@gerard33 i've tagged a new release that should support your `myDeviceArray`, let me know if it works :)

https://github.com/robbinjanssen/home-assistant-omnik-inverter/releases/tag/v0.0.3

closed 2020-02-24 by robbinjanssen

gerard33, 2020-02-24:
Great. That works for me indeed :+1: