robbinjanssen/home-assistant-omnik-inverter #13: repositroy.json missing

jbr79, 2020-04-29:
When adding the repository in HACS, the log shows the following error:

20-04-29 07:22:04 ERROR (MainThread) [supervisor.utils.json] Can't read json from /data/addons/git/055ed765/repository.json: [Errno 2] No such file or directory: '/data/addons/git/055ed765/repository.json'
20-04-29 07:22:04 WARNING (MainThread) [supervisor.store.data] Can't read repository information from /data/addons/git/055ed765/repository.json

robbinjanssen, 2020-04-29:
I think you are not adding it through HACS but you are adding it through hassio?

You need to add it via `/hacs/settings`

jbr79, 2020-04-30:
ah for crying out loud... I need some time off.

Works!

closed 2020-04-30 by jbr79

robbinjanssen, 2020-05-01:
Great to hear you resolved it, actually created a PR in HACS now to add it as a default repo :-)

robbinjanssen, 2020-05-01:
To prevent this in the future, we've now been added as an official integration of HACS 🎉 

https://github.com/hacs/default/pull/383