robbinjanssen/home-assistant-omnik-inverter #15: No data found

GoSpursGoNL, 2020-05-03:
Thanks for this promising addon!
For me unfortunately it doesn't work yet, hope you can help.

The addon logs:
Empty data from Omnik Inverter 192.168.1.28

Config is as follows:
`  - platform: omnik_inverter
    host: 192.168.1.28
[image]
`

http://192.168.1.28/js/status.js returns:
`var txt;var opt;function initPage(){txt=window.parent.reList().page.status;opt=window.parent.reList().page.option;for(var A=1;A<13;A++){inner("t"+A,txt["t"+A]);inner("p1_"+A,txt.pg1[A]);inner("p2_"+A,txt.pg2[A]);inner("p3_"+A,txt.pg3[A]);inner("p4_"+A,txt.pg4[A])}inner("p3_1_1",txt.pg1[6]);inner("p3_2_1",txt.pg1[6]);inner("p3_1_2",txt.pg3[6]);inner("p3_2_2",txt.pg3[6]);inner("p3_1_3",txt.pg3[4]);inner("p3_2_3",txt.pg3[4]);inner("p3_2_4",txt.pg3[5]);inner("p3_2_5",txt.pg3[7]);changeFont();child_getH()}function initData(){child_getH();ajax("status.json?CMD=inv_query&rand="+Math.random())}function ajaxReturn(B){if(B.time!=null){inner("gtype",txt.pg1["s3"]);inner("gsn",B.g_sn);inner("ver",B.g_ver);inner("time",new Date(B.time*1000).toString());inner("worktime",B.g_time+" "+txt.pg1["s4"]);inner("error",B.g_err);inner("mac",B.mac);inner("auto_ip",enb_txt(B.auto_ip));inner("ip",B.ip);inner("sub",B.sub);inner("gate",B.gate);inner("auto_dns",enb_txt(B.auto_dns));inner("dns",B.dns);inner("dns_bak",B.dns_bak);inner("type",reInvBrand(B.i_type));if(B.i_num==null||B.i_num==0){}else{show("inv_data",true);inner("list",B.i_sn);var A=[B.i_sn,B.i_ver_m,B.i_ver_s,B.i_modle,B.i_pow,B.i_pow_n,B.i_eday,B.i_eall,B.i_alarm,B.i_last_t];inv_data_show(A)}inner("ip_a",B.l_a_ip);inner("port_a",B.l_a_port);inner("agreement_a",agr_txt(9,B.l_a_agr));if(B.ip_b!=null){show("server_b",true);inner("ip_b",B.ip_b);inner("port_b",B.port_b);inner("domain_b",B.ipdm_b);inner("agreement_b",agr_txt(9,B.agr_b))}if(B.i_status!=null){if(B.i_status==0){show("inv_data",false)}else{show("inv_data",true)}}}if(B.conn_a!=null){initStatus("status_0",B.conn_a)}if(B.conn_b!=null){initStatus("status_1",B.conn_b)}}var sels=1;function upfold(A){var B=getCon("pg"+A);if(B.style.display=="none"){B.style.display="";getCon("p"+A).src="image/up.png"}else{B.style.display="none";getCon("p"+A).src="image/down.png"}child_getH();if(A==3&&B.style.display!="none"){ajax("s_ping.json?CMD=inv_query&server=110&rand="+Math.random())}}function inv_data_show(A){var D=new Array("list","mver","sver","inv_tp","rate_p","now_p","day_e","all_e","alarm","uptime");if(A!=null){for(var B=1;B<10;B++){if(A[B]==""){if(isNaN(parseInt(A[B]))){inner(D[B],"---");continue}}var C=A[B];switch(B){case 4:C+=" W";break;case 5:if(C*1==4294967295){C="---"}else{C+=" W"}break;case 6:if(C*1==4294967295){C="---"}else{C+=" kWh"}break;case 7:if(C*1==4294967295){C="---"}else{C+=" kWh"}break;case 9:if(getCon("list").innerHTML=="---"){C="---"}else{C+=txt.pg2["s2"]}break}inner(D[B],C)}}}function initStatus(C,B){var A=txt.pg3["s1"];var D=txt.pg3["s2"];if(B==1){getCon(C).innerHTML=A}else{getCon(C).innerHTML=D}}function enb_txt(A){var B=txt.pg2["s4"];if(A==1){B=txt.pg2["s3"]}return B}function agr_txt(C,A){var B="";if(C==1){B=opt[1];if(A==1){B=opt[2]}}if(C==3){B=opt[3];if(A==1){B=opt[4]}}if(C==5){B=opt[5];if(A==1){B=opt[6]}}if(C==7){B=opt[7];if(A==1){B=opt[8]}}if(C==9){if(A==2){B=opt[11]}else{if(A==1){B=opt[10]}else{B=opt[9]}}}return B};`

If I open manually http://192.168.1.28 I see the data, please see image.

robbinjanssen, 2020-05-04:
Interesting, your JS files returns completely different data then mine. I see you use an omnik 2k, i've only tested this with an omnik 4k.

When beautifying the code it looks like your inverter is calling some other URL to get the data, can you try calling `http://192.168.1.28/status.json?CMD=inv_query&rand=1234` and see if it returns anything?

If so we might be able to add support for this inverter as well :-)

```js
var txt;
var opt;

function initPage() {
    txt = window.parent.reList().page.status;
    opt = window.parent.reList().page.option;
    for (var A = 1; A < 13; A++) {
        inner("t" + A, txt["t" + A]);
        inner("p1_" + A, txt.pg1[A]);
        inner("p2_" + A, txt.pg2[A]);
        inner("p3_" + A, txt.pg3[A]);
        inner("p4_" + A, txt.pg4[A])
    }
    inner("p3_1_1", txt.pg1[6]);
    inner("p3_2_1", txt.pg1[6]);
    inner("p3_1_2", txt.pg3[6]);
    inner("p3_2_2", txt.pg3[6]);
    inner("p3_1_3", txt.pg3[4]);
    inner("p3_2_3", txt.pg3[4]);
    inner("p3_2_4", txt.pg3[5]);
    inner("p3_2_5", txt.pg3[7]);
    changeFont();
    child_getH()
}

function initData() {
    child_getH();
    ajax("status.json?CMD=inv_query&rand=" + Math.random())
}

function ajaxReturn(B) {
    if (B.time != null) {
        inner("gtype", txt.pg1["s3"]);
        inner("gsn", B.g_sn);
        inner("ver", B.g_ver);
        inner("time", new Date(B.time * 1000).toString());
        inner("worktime", B.g_time + " " + txt.pg1["s4"]);
        inner("error", B.g_err);
        inner("mac", B.mac);
        inner("auto_ip", enb_txt(B.auto_ip));
        inner("ip", B.ip);
        inner("sub", B.sub);
        inner("gate", B.gate);
        inner("auto_dns", enb_txt(B.auto_dns));
        inner("dns", B.dns);
        inner("dns_bak", B.dns_bak);
        inner("type", reInvBrand(B.i_type));
        if (B.i_num == null || B.i_num == 0) {} else {
            show("inv_data", true);
            inner("list", B.i_sn);
            var A = [B.i_sn, B.i_ver_m, B.i_ver_s, B.i_modle, B.i_pow, B.i_pow_n, B.i_eday, B.i_eall, B.i_alarm, B.i_last_t];
            inv_data_show(A)
        }
        inner("ip_a", B.l_a_ip);
        inner("port_a", B.l_a_port);
        inner("agreement_a", agr_txt(9, B.l_a_agr));
        if (B.ip_b != null) {
            show("server_b", true);
            inner("ip_b", B.ip_b);
            inner("port_b", B.port_b);
            inner("domain_b", B.ipdm_b);
            inner("agreement_b", agr_txt(9, B.agr_b))
        }
        if (B.i_status != null) {
            if (B.i_status == 0) {
                show("inv_data", false)
            } else {
                show("inv_data", true)
            }
        }
    }
    if (B.conn_a != null) {
        initStatus("status_0", B.conn_a)
    }
    if (B.conn_b != null) {
        initStatus("status_1", B.conn_b)
    }
}
var sels = 1;

function upfold(A) {
    var B = getCon("pg" + A);
    if (B.style.display == "none") {
        B.style.display = "";
        getCon("p" + A).src = "image/up.png"
    } else {
        B.style.display = "none";
        getCon("p" + A).src = "image/down.png"
    }
    child_getH();
    if (A == 3 && B.style.display != "none") {
        ajax("s_ping.json?CMD=inv_query&server=110&rand=" + Math.random())
    }
}

function inv_data_show(A) {
    var D = new Array("list", "mver", "sver", "inv_tp", "rate_p", "now_p", "day_e", "all_e", "alarm", "uptime");
    if (A != null) {
        for (var B = 1; B < 10; B++) {
            if (A[B] == "") {
                if (isNaN(parseInt(A[B]))) {
                    inner(D[B], "---");
                    continue
                }
            }
            var C = A[B];
            switch (B) {
                case 4:
                    C += " W";
                    break;
                case 5:
                    if (C * 1 == 4294967295) {
                        C = "---"
                    } else {
                        C += " W"
                    }
                    break;
                case 6:
                    if (C * 1 == 4294967295) {
                        C = "---"
                    } else {
                        C += " kWh"
                    }
                    break;
                case 7:
                    if (C * 1 == 4294967295) {
                        C = "---"
                    } else {
                        C += " kWh"
                    }
                    break;
                case 9:
                    if (getCon("list").innerHTML == "---") {
                        C = "---"
                    } else {
                        C += txt.pg2["s2"]
                    }
                    break
            }
            inner(D[B], C)
        }
    }
}

function initStatus(C, B) {
    var A = txt.pg3["s1"];
    var D = txt.pg3["s2"];
    if (B == 1) {
        getCon(C).innerHTML = A
    } else {
        getCon(C).innerHTML = D
    }
}

function enb_txt(A) {
    var B = txt.pg2["s4"];
    if (A == 1) {
        B = txt.pg2["s3"]
    }
    return B
}

function agr_txt(C, A) {
    var B = "";
    if (C == 1) {
        B = opt[1];
        if (A == 1) {
            B = opt[2]
        }
    }
    if (C == 3) {
        B = opt[3];
        if (A == 1) {
            B = opt[4]
        }
    }
    if (C == 5) {
        B = opt[5];
        if (A == 1) {
            B = opt[6]
        }
    }
    if (C == 7) {
        B = opt[7];
        if (A == 1) {
            B = opt[8]
        }
    }
    if (C == 9) {
        if (A == 2) {
            B = opt[11]
        } else {
            if (A == 1) {
                B = opt[10]
            } else {
                B = opt[9]
            }
        }
    }
    return B
};
```

GoSpursGoNL, 2020-05-04:
Thanks man, yes that returns the data. I redacted g_sn and mac values.
First url:
`{"g_sn":xxx,"g_ver":"VER:ME-111001-V1.0.6(2015-10-16)","time":1588598399,"g_time":8002,"g_err":0,"mac":"xxx","auto_ip":1,"ip":"192.168.1.28","sub":"255.255.255.0","gate":"192.168.1.9","auto_dns":1,"dns":"8.8.8.8","dns_bak":"8.8.4.4","i_status":16,"i_type":258,"i_num":1,"i_sn":"xxx","i_ver_m":"V1.25Build23261","i_ver_s":"V1.40Build52927","i_modle":"omnik2000tl2","i_pow":"2000","i_pow_n":1225,"i_eday":"10.90","i_eall":"8674.0","i_alarm":"F23","i_last_t":0,"a_agr":0,"l_a_ip":"47.88.8.200","l_a_port":10000}`

i_pow_n = current AC power in watt
i_eday = energy today in kWh
i_eall = total energy

Second url returns:
`
{"conn_a":1}
`

robbinjanssen, 2020-05-05:
Awesome, I think we can work with the first URL. I'll try creating a test version for you later this week.

robbinjanssen, 2020-05-06:
@GoSpursGoNL is `"i_eall":"8674.0",` kWh ?

robbinjanssen, 2020-05-06:
I've created a test version, unfortunately I cannot test this myself. Do you mind testing this for me?

https://github.com/robbinjanssen/home-assistant-omnik-inverter/tree/rja/v1.3.0

1. Copy the sensor.py file from that branch to your home assistant
2. Add `use_json: true` in your configuration.yaml

GoSpursGoNL, 2020-05-06:
Yes, in kWh.
Config like this?
```
sensor:
  - platform: omnik_inverter
    host: 192.168.1.28
    use_json: true
```
I replaced the sensor.py, have not restarted HA yet, made the config changes as per above.
But config validation says:
Invalid config for [sensor.omnik_inverter]: [use_json] is an invalid option for [sensor.omnik_inverter]. Check: sensor.omnik_inverter->use_json. (See ?, line ?).

robbinjanssen, 2020-05-06:
@GoSpursGoNL I had an error in the sensor.py, fixed it now. Also tagged it as a `pre-release` so if you have beta releases enabled in HACS then you can add/update it through there. Your config looks fine, and i'm sure that HA won't bug you about it this time :)

GoSpursGoNL, 2020-05-07:
It works, awesome! Since this morning I did two additional restarts of HA but I now notice that additional entities were created. So the original ones are now unavailable and three new ones with an "_2" added to the entity id were created that have data. Can this be avoided?

robbinjanssen, 2020-05-07:
Glad it works! I ran into the same problem, I think "copy pasting" the sensor is not the way to go in HACS.

I got it working by:
1. Remove the integration via HACS
2. (might need a restart in between) Remove all entities via http://home.assistant:8123/config/entities (search for `solar`)
3. Restart home assistant
4. Re-add the integration via HACS

I'll create a proper release for this version now 👍

robbinjanssen, 2020-05-07:
I made an error in the sensor.py, I think that is causing the `solar_*_2` bug. I'll try to fix it

robbinjanssen, 2020-05-07:
@GoSpursGoNL please update to v1.3.1, that should fix the problem. You might want to delete all the entities again by repeating the steps above.

PR #19 (2020-05-07): Make unique identifier work again
### What does it do?
`self._unique_id = f"{self.data.result[0]}-{self._name}"` no longer worker because result[0] contains the "current" power, so the unique_id would change everytime.

Now result[0] matches the ID of the inverter.

Fixes #15 (again 😂)
closed 2020-05-07 by robbinjanssen