robbinjanssen/home-assistant-omnik-inverter #16: Empty data from Omnik Inverter 192.168.0.58 - But Work

wes1993, 2020-05-05:
Hello, Thanks a lot for your beautiful Project :-D
Works really well.
just one thing sometimes in my HomeAssistant log i have this error log:
Empty data from Omnik Inverter 192.168.0.58
But i can see the kwh etc in homeassistant so i think that the addon work because can take the info from my inverter.
can i send to you some stuff to identify why sometimes won't work?
Thanks a lot
Stefano

robbinjanssen, 2020-05-05:
Hi @wes1993, sure! I'll need some more info;

1. What does your inverter output if you go to http://192.168.0.58/js/status.js? 
2. Is it within specific timeframes when it doesn't work?

The [code](https://github.com/robbinjanssen/home-assistant-omnik-inverter/blob/master/custom_components/omnik_inverter/sensor.py#L90-L100) that outputs the error basically tries to find the data within the string that is outputted by the URL above, if it cannot find any match it outputs the error.

To see what the inverter returns when the "empty data" error is outputted you need to enable logging on debug level, add the following to your configuration.yaml:

```yaml
# Example configuration.yaml entry
logger:
  default: warning
  logs:
    custom_components.omnik_inverter: debug
```

wes1993, 2020-05-05:
The response from the status.js:

var version= "H4.01.51MW.2.01W1.0.64(2018-01-251-D)";
var m2mMid = "Hidden";
var wlanMac= "Hidden";
var m2mRssi= "65%";
var wanIp= "192.168.0.58";
var nmac= "Hidden";
var fephy= "off";
var yz_device_type="258";
var yz_device_num="1";
var myDeviceArray=new Array();
myDeviceArray[0]="IELN1520178F5210,V4.08Build215,V4.12Build246,Omnik1500tl ,1500,942,552,24693,,1,";
;
function initPageText()
{
    var list = window.parent.reList("status");
    fileText("st1",list["t1"]);
    fileText("st2",list["t2"]);
    fileText("st3",list["t3"]);
    fileText("st4",list["t4"]);
    for(var i=1;i<=32;i++)
    {
        if(i!=14)
        {
            fileText("tx"+i,list[i]);
        }
    }
    init_main_page();
    initStatus();
    initInvList(parseInt(yz_device_type),yz_device_num);
    changeFont();
    child_getH();
}
function upfold(v)
{
    if(document.getElementById("up_"+v+"_div").style.display == "none")
    {
        show("up_"+v+"_div");
        document.getElementById("up_"+v).src = "image/up.png";
    }
    else
    {
        hide("up_"+v+"_div");
        document.getElementById("up_"+v).src = "image/down.png";
    }
    if(v == 3)
    {
        if(document.getElementById("ser_status").src=="")
        {
			//alert("mhtest");
            document.getElementById("ser_status").src="server_status.html"; 
        }
    }
}
function initInvList(type,num)
{
    if(type==null || type == "")
    {
        type = window.parent.reTip(40);
    }
    else
    {
        type = reInvBrand(type);
    }
    if(num==null || num == "")
    {
        hide("inv_msg_div");
        num = window.parent.reTip(40);
    }
[... 192 lines cut ...]
		    if(value*1==4294967295)
		    {
		        elem.innerHTML = "---";
		    }
		    else
		    {
		        sv= value*1/10;
			    elem.innerHTML= sv.toString()+" kWh";
		    }
		}
		else if (nv== 10)
		{
			if (document.getElementById("webdata_sn").innerHTML== "---")
				elem.innerHTML = "---";
			else
				elem.innerHTML = value+window.parent.reTip("5");
		}
		else
			elem.innerHTML = value;
	}
	else
	{
		alert(webdata_ids[nv]);
	}
}

wes1993, 2020-05-05:
The error log:

Logger: homeassistant.helpers.entity
Source: __main__.py:347
First occurred: 13:45:37 (1 occurrences)
Last logged: 13:45:37
Update of sensor.solar_power_current is taking over 10 seconds

Logger: custom_components.omnik_inverter.sensor
Source: custom_components/omnik_inverter/sensor.py:146
Integration: omnik_inverter (documentation)
First occurred: 10:42:35 (10 occurrences)
Last logged: 14:27:09

Empty data from Omnik Inverter 192.168.0.58
Unable to fetch data from Omnik Inverter 192.168.0.58

robbinjanssen, 2020-05-05:
Hm that all seems normal, if you "sometimes" get data it means the output can be parsed and analysed. Maybe your inverter looses the network connection sometimes? That would explain the "Unable to fetch data from Omnik Inverter 192.168.0.58" message

Can you open a terminal window and `ping 192.168.0.58` for a while? See if there are some connection drops or something?

I'm working on an update for the inverter that reduces the nr of calls to the inverter (once every 5 minutes instead of once every minute), maybe that helps as well.

wes1993, 2020-05-05:
Thanks for your fast reply :-D
i will try to ping for a while.
last question can you add the alert sensor and fatch the data from the java response?
Thanks a lot
Stefano

wes1993, 2020-05-05:
This is my log:

2020-05-05 15:20:04 DEBUG (SyncWorker_14) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '868', '622', '24700', '', '2', '']
2020-05-05 15:21:35 DEBUG (SyncWorker_11) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '862', '625', '24700', '', '1', '']
2020-05-05 15:22:39 ERROR (SyncWorker_11) [custom_components.omnik_inverter.sensor] Empty data from Omnik Inverter 192.168.0.58
2020-05-05 15:22:39 DEBUG (SyncWorker_11) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '862', '625', '24700', '', '1', '']
2020-05-05 15:24:09 DEBUG (SyncWorker_6) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '873', '631', '24701', '', '0', '']
2020-05-05 15:25:13 DEBUG (SyncWorker_6) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '873', '631', '24701', '', '1', '']
2020-05-05 15:26:44 DEBUG (SyncWorker_1) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '886', '634', '24701', '', '0', '']
2020-05-05 15:27:48 DEBUG (SyncWorker_5) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '886', '634', '24701', '', '1', '']
2020-05-05 15:29:19 DEBUG (SyncWorker_19) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '897', '637', '24701', '', '0', '']
2020-05-05 15:30:23 DEBUG (SyncWorker_9) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '897', '637', '24701', '', '1', '']
2020-05-05 15:31:54 DEBUG (SyncWorker_13) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '885', '641', '24702', '', '0', '']
2020-05-05 15:32:58 DEBUG (SyncWorker_12) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '885', '641', '24702', '', '1', '']
2020-05-05 15:34:29 DEBUG (SyncWorker_9) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '873', '644', '24702', '', '1', '']
2020-05-05 15:35:32 DEBUG (SyncWorker_10) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '873', '644', '24702', '', '2', '']
2020-05-05 15:37:03 ERROR (SyncWorker_4) [custom_components.omnik_inverter.sensor] Empty data from Omnik Inverter 192.168.0.58
2020-05-05 15:37:03 DEBUG (SyncWorker_4) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '873', '644', '24702', '', '2', '']
2020-05-05 15:38:04 DEBUG (SyncWorker_3) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '845', '650', '24703', '', '1', '']
2020-05-05 15:39:06 DEBUG (SyncWorker_13) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '845', '650', '24703', '', '2', '']
2020-05-05 15:40:07 DEBUG (SyncWorker_17) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '825', '653', '24703', '', '0', '']
2020-05-05 15:41:09 DEBUG (SyncWorker_3) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '825', '653', '24703', '', '1', '']
2020-05-05 15:42:13 DEBUG (SyncWorker_5) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '823', '657', '24703', '', '0', '']
2020-05-05 15:43:44 DEBUG (SyncWorker_2) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '823', '657', '24703', '', '1', '']
2020-05-05 15:44:46 DEBUG (SyncWorker_10) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '814', '660', '24704', '', '0', '']
2020-05-05 15:45:48 DEBUG (SyncWorker_19) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '814', '660', '24704', '', '1', '']
2020-05-05 15:46:50 DEBUG (SyncWorker_9) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '795', '664', '24704', '', '0', '']
2020-05-05 15:47:52 DEBUG (SyncWorker_5) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '795', '664', '24704', '', '1', '']
2020-05-05 15:48:56 DEBUG (SyncWorker_14) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '795', '664', '24704', '', '2', '']
[... cut ...] 8Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '781', '666', '24704', '', '1', '']
2020-05-05 15:51:30 DEBUG (SyncWorker_16) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '781', '666', '24704', '', '2', '']
2020-05-05 15:53:00 DEBUG (SyncWorker_10) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '787', '669', '24704', '', '1', '']

robbinjanssen, 2020-05-06:
Strange, it does seem like the script gets data, @gerard33 do you have any idea?

```
2020-05-05 15:22:39 ERROR (SyncWorker_11) [custom_components.omnik_inverter.sensor] Empty data from Omnik Inverter 192.168.0.58
2020-05-05 15:22:39 DEBUG (SyncWorker_11) [custom_components.omnik_inverter.sensor] Data = ['IELN1520178F5210', 'V4.08Build215', 'V4.12Build246', 'Omnik1500tl ', '1500', '862', '625', '24700', '', '1', '']
```

Looking at the timestamps the data is actually there and the script does output it 🤔 

```py
# Find the webData
if result.find('webData="') != -1:
    matches = re.search(r'(?<=webData=").*?(?=";)', result)
else:
    matches = re.search(r'(?<=myDeviceArray\[0\]=").*?(?=";)', result)

# Split the values
if matches is not None:
    self.result = matches.group(0).split(',')
else:
    _LOGGER.error("Empty data from Omnik Inverter %s", self._host)

_LOGGER.debug("Data = %s", self.result)
```

@wes for your other question, what do you mean with the "alert sensor"?

wes1993, 2020-05-07:
Perfect for the ping :-D
I think there are another option that is called Alert.
if we can take this option and make a sensor in HA

Thanks a Lot
Stefano

robbinjanssen, 2020-05-07:
I'm sorry but I don't know what that option is or where I can find it. It's not in the status.js.

If you want to create an alert when the power hits a specific kWh then you can create an automation in home assistant.

Please try out the new version (1.3.0) and see if your data is more stable 👍

wes1993, 2020-05-07:
Hello Again :-D i have updated the addon, i will update you  :-D
if i open the status page of the inverter i have this string:

[image]

and i this is this value:
myDeviceArray[0]="IELN1520178F5210,V4.08Build215,V4.12Build246,Omnik1500tl ,1500,482,851,24883,here there is a blank space,0,";

myDeviceArray[0]="IELN1520178F5210,V4.08Build215,V4.12Build246,Omnik1500tl ,1500,482,851,24883,,0,";

so we can make a sensor if = "" then no error else error something like this, but my inverter never has the alert turned on so i don't know what value will write if there is some error.

Thanks a Lot
Stefano

robbinjanssen, 2020-05-07:
Ah now I understand what you are talking about, we could do something with that, but then we need to know what kind of "errors" appear there. Could be a string describing the error or a code "F1234", we don't know as of right now. If you can find any information about that I might be able to fix something.

What we could do is create a "boolean" that simply says "error: yes/no"

wes1993, 2020-05-15:
Hello Again Robbin, for now i have the same problem... :-(
i don't know why..
[image]

robbinjanssen, 2020-05-19:
@wes1993 i'll add some more logging to the empty data error when I have the time, we might see some more info then. I'll update this issue when i've done this :)

wes1993, 2020-05-19:
Perfect!! :) thanks a lot :)

robbinjanssen, 2020-08-04:
@wes1993 are you still experiencing this? If so please reopen :)

closed 2020-08-04 by robbinjanssen

robbinjanssen, 2020-08-05:
@wes1993 i've opened a separate issue for this https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/27 :)