robbinjanssen/home-assistant-omnik-inverter #20: No output with js or json.

florafrisia, 2020-05-11:
Hello,

I have two onmik convertors in my house. The newest convertor is working perfectly and the oldest one has an issue. After I run the javascript I have to log in and after the logging in nothing happens. I get only Access Error: Site or Page Not Found. Can you assist me with this. With json I don't have to login but the result is the same.

Regards,

Peter

robbinjanssen, 2020-05-12:
Hi @florafrisia, It could be that your omnik inverter (or rather specific the wifi chip) inside the inverter is not supported. What kind of inverter do you have? Omnik 3?

florafrisia, 2020-05-12:
Hello robbin,

I have a 3k tl with first version of the wifi module. After some research version 2 till 6 you can extract data from it and not from the first one. So when I saw this project let give it a try. Sadly it will not work I think. 

Regards,

Peter

robbinjanssen, 2020-05-12:
@florafrisia, I'm afraid you're right and it won't work. You could do some research in your network tab and see if the web interface (after logging in) is loading some javascript files which output the inverter info. If so then we might be able to extract it using http://user:pass@192.168.100.100/js/whatever-the-file-is.js

robbinjanssen, 2020-05-25:
Closing due to inactivity

closed 2020-05-25 by robbinjanssen