robbinjanssen/home-assistant-omnik-inverter #21: Feature request: set update interval

silvanverschuur, 2020-05-14:
Great project! I've done the same using a Bash script that parses the web page of the Omnik inverter and publishes the result to MQTT.

Because I like your approach (custom component) I would like to use this HA integration instead of my own Bash script. The only thing I'm missing is a variable update interval. At the moment I'm executing my Bash script every minute. I've read that your component updates the state every 5 minutes (in your code it seem 500 seconds).

Is it possible to move this hard-coded interval to some kind of configurable setting?

robbinjanssen, 2020-05-15:
Whoops, those 500 should be 300, my bad! But making in configurable seems a good idea, i'll see what I can do :)

robbinjanssen, 2020-05-15:
@silvanverschuur I've just released v1.3.2 making the update time the 5 minutes it should be. I'll keep this issue open to make it configurable 👍

silvanverschuur, 2020-05-15:
That would be great!

robbinjanssen, 2020-05-17:
All done!

Try out 1.4.0 and set a `scan_interval: 60` in your configuration.yaml. (Time is in seconds)

silvanverschuur, 2020-05-17:
@robinjanssen Thank you for implementing this feature! 

I just installed version 1.4.0. When I set the scan_interval HA gives the following error:

`Invalid config for [sensor.omnik_inverter]: Expected seconds, got 0:15:00 for dictionary value @ data['scan_interval']. Got datetime.timedelta(seconds=900). (See ?, line ?). `

robbinjanssen, 2020-05-17:
What did you use for `scan_interval` value?

silvanverschuur, 2020-05-17:
I tried to set it to 60, but I also tried the default `scan_interval: 900`

Both values give the same error

robbinjanssen, 2020-05-17:
Strange, i've set mine to 30 now and it seems to work:

```
config $ ha core log | grep omnik
2020-05-17 15:56:07 INFO (SyncWorker_2) [homeassistant.loader] Loaded omnik_inverter from custom_components.omnik_inverter
2020-05-17 15:56:14 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for omnik_inverter which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant.
2020-05-17 15:56:14 DEBUG (SyncWorker_6) [custom_components.omnik_inverter.sensor] Data = ['xxx', 952, 630, 95163]
2020-05-17 15:56:46 DEBUG (SyncWorker_0) [custom_components.omnik_inverter.sensor] Data = ['xxx', 952, 630, 95163]
2020-05-17 15:57:19 DEBUG (SyncWorker_12) [custom_components.omnik_inverter.sensor] Data = ['xxx', 952, 630, 95163]
2020-05-17 15:58:19 DEBUG (SyncWorker_15) [custom_components.omnik_inverter.sensor] Data = ['xxx', 952, 630, 95163]
2020-05-17 15:58:50 DEBUG (SyncWorker_17) [custom_components.omnik_inverter.sensor] Data = ['xxx', 1022, 639, 95163]
```

What version of home assistant do you have?

robbinjanssen, 2020-05-17:
Aha, i see what you mean now. The config validation fails. Let me look into that

silvanverschuur, 2020-05-17:
I'm using Home Assistant version 0.109.6. Could it be a config issue? I'm using a separate config file for each sensor:

`sensor: !include_dir_list ../entities/sensors`

And I've added a file `omnik.yaml` with: 

```
# https://github.com/robbinjanssen/home-assistant-omnik-inverter
platform: omnik_inverter
host: 192.168.1.18
scan_interval: 900
```

robbinjanssen, 2020-05-17:
It's something in the code validation config, looking into it. Quite new to this as well so unfortunately there's a lot of trial and error involved 🤦

robbinjanssen, 2020-05-17:
Found it, v1.4.1 coming up

robbinjanssen, 2020-05-17:
Done, all tagged!

silvanverschuur, 2020-05-17:
@robbinjanssen Thanks for the quick fix. I can confirm that this fixes the config validation error.
When I restart HA I now get the following error:

```
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.7/site-packages/homeassistant/helpers/entity_platform.py", line 178, in _async_setup_platform
    await asyncio.wait_for(asyncio.shield(task), SLOW_SETUP_MAX_WAIT)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/homeassistant/.homeassistant/custom_components/omnik_inverter/sensor.py", line 73, in setup_platform
    entities.append(OmnikInverterSensor(data, sensor_type, cache))
  File "/home/homeassistant/.homeassistant/custom_components/omnik_inverter/sensor.py", line 174, in __init__
    self.update()
  File "/home/homeassistant/.homeassistant/custom_components/omnik_inverter/sensor.py", line 264, in update
    pickle.dump([cacheValue, cacheDay], open(cacheName, 'wb'))
PermissionError: [Errno 13] Permission denied: '.powertoday.pickle'
```

I disabled the cache for now. Could it be that the following line in your source code is invalid?

`cache = pickle.load(open(cacheName, 'rb'))`

It seems to me that the file is opened read-only.

robbinjanssen, 2020-05-17:
Glad to hear it is working!

Hmm that could be the case indeed, i’ll have to look into this. I’ll update when i know more

robbinjanssen, 2020-05-18:
@silvanverschuur This might be just a permission problem on your system. Do you run hassio? Or a custom installation?

Maybe try deleting the file and then see if it can write the new file? All my files belong to `root` in the config folder

`-rw-r--r--    1 root     root            15 May 18 09:16 .powertoday.pickle`

silvanverschuur, 2020-05-18:
@robbinjanssen I'm running Home Assistant in a Python virtual environment. My home assistant user is owner of the whole `.homeassistant` directory. I checked the permissions of the sub folders but everything seems ok. Other integrations are capable of adding / changing files inside my `.homeassistant` directory (e.g. Google, PS4, MySensors integrations).

Where is the `.powertoday.pickle` file located? In the same folder as your `configuration.yaml` (config root)?

robbinjanssen, 2020-05-19:
Yes it should be right there

robbinjanssen, 2020-05-19:
I see that the PS4 components utilises home assistant functionality: https://github.com/home-assistant/core/blob/dev/homeassistant/components/ps4/__init__.py#L160-L185

When I have the time i'll rewrite the "caching" to a json file and load/save it like the component above.

silvanverschuur, 2020-05-19:
@robbinjanssen Take your time. As far as I can see right now, I don't need caching at all because my Omnik inverter doesn't reset the daily total.

Your plugin works great, thanks!

robbinjanssen, 2020-08-04:
@silvanverschuur v1.5.0 uses the HASS helpers to define the storage path and now uses JSON instead of pickle to save the "cached" status :)

PR #23 (2020-05-17): Make scan_interval configurable
### What does it do?
Add a `scan_interval` configuration option to configure the time between updates

Closes #21
closed 2020-08-04 by robbinjanssen