robbinjanssen/home-assistant-omnik-inverter #26: Omnik is not working in HA PermissionError: [Errno 13] Permission denied: '.powertoday.pickle'

florafrisia, 2020-06-18:
I get the following error:

```
2020-06-18 20:57:03 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up omnik_inverter platform for sensor
Traceback (most recent call last):
  File "/home/ha/homeassistant/lib/python3.7/site-packages/homeassistant/helpers/entity_platform.py", line 178, in _async_setup_platform
    await asyncio.wait_for(asyncio.shield(task), SLOW_SETUP_MAX_WAIT)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 442, in wait_for
    return fut.result()
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/ha/homeassistant/custom_components/omnik_inverter/sensor.py", line 73, in setup_platform
    entities.append(OmnikInverterSensor(data, sensor_type, cache))
  File "/home/ha/homeassistant/custom_components/omnik_inverter/sensor.py", line 174, in __init__
    self.update()
  File "/home/ha/homeassistant/custom_components/omnik_inverter/sensor.py", line 264, in update
    pickle.dump([cacheValue, cacheDay], open(cacheName, 'wb'))
PermissionError: [Errno 13] Permission denied: '.powertoday.pickle'

```

florafrisia, 2020-06-18:
Can you help me with this error ?

Regards,

Peter

robbinjanssen, 2020-06-19:
Hi @florafrisia, Yes i'm planning to update the "cache" implementation to something more in lines with home assistant. I'll update this issue when I do :)

In the mean time you could try disabling the cache by adding `cache_power_today: false` to your configuration.

Thanks for reporting!

robbinjanssen, 2020-08-04:
@florafrisia v1.5.0 uses the HASS helpers to define the storage path and now uses JSON instead of pickle to save the "cached" status :)

I think this will work for you as well :)

closed 2020-08-04 by robbinjanssen