robbinjanssen/home-assistant-omnik-inverter #28: Use of requests

gerard33, 2020-12-17:
Hi Robbin, I have switched over from this custom integration to an Appdaemon script to get the data locally from my inverter. Mostly because that way I can use the same moment on which the data from the inverter and from DSMR is updated. One of the advantages of Appdaemon is that you can change code and directly use it, without restarting HA.
I re-used a lot of your code in the Appdaemon app. Recently I switched from `urlopen` to `requests` and to me that seems to work more stable. The number of errors in the log has dropped a lot.

There are e.g. no errors with `fp.close()` when there is no data and also I am using a timeout.
The change to `requests` should be rather simple. If you are interested I can share my code so you can make a separate branch to test that version. Let me know your thoughts.

robbinjanssen, 2021-01-24:
Sorry for the late response. I like the idea and will happily look at your code. Since you've already done a lot for this component I could also make you co-owner of this repository? That way you can create your own branch and we could together implement and improve this?

gerard33, 2021-01-26:
I will share the relevant part of the script here so you can see which parts you can re-use. No need to add me as co-owner as I will keep using the Appdaemon script and have to much on my to-do list so wont have time to further test this as a custom integration 😄

The script detects if no connection can be made and in that case restarts. That was the only way I could get data from the inverter again. Let me know if there are any questions regarding the script.

gerard33, 2021-01-26:
```py
"""Omnik inverter solar AppDaemon app."""
import appdaemon.plugins.hass.hassapi as hass

from datetime import datetime, timedelta

import requests

SOLAR_SENSOR_TYPES = {
    "solar_power_current": {
        "friendly_name": "Zonne-energie huidig",
        "unit_of_measurement": "W",
        "icon": "mdi:solar-power",
    },
    "solar_power_current_kw": {
        "friendly_name": "Zonne-energie vandaag kW",
        "unit_of_measurement": "kW",
        "icon": "mdi:solar-power",
    },
    "solar_power_today": {
        "friendly_name": "Zonne-energie vandaag",
        "unit_of_measurement": "kWh",
        "icon": "mdi:solar-power",
    },
    "solar_power_total": {
        "friendly_name": "Zonne-energie totaal",
        "unit_of_measurement": "kWh",
        "icon": "mdi:solar-power",
    },
    "solar_wifi_strength": {
        "friendly_name": "Zonnepanelen Wifi Sterkte",
        "unit_of_measurement": "%",
        "icon": "mdi:wifi",
    },
}

class SolarReader(hass.Hass):
    """Solar class."""

    def initialize(self):
        """Call Initialize Solar
        Used this way so app can reset it self when there is an error reading the Omnik inverter.
        """
        self.real_initialization()

    def real_initialization(self, message_sent=False):
        """Initialize Solar."""
        self.log("Solar App started.")
        self.ip_address = self.args.get("ip_address")
        self.interval = self.args.get("interval", 30)
        self.chat_id = self.args.get("chat_id")
        self.title = self.args.get("title")
        self.debug = self.args.get("debug", False)
        self.result = None
        self.error_counter = 0
        self.message_sent = message_sent

        self.run_every(self.get_solar_data, "now", self.interval)

    def get_solar_data(self, kwargs):
        """Get the data from the Omnik inverter."""
        dataurl = f"http://{self.ip_address}/js/status.js"

        if self.debug:
            self.log(dataurl, log="solar_log", level="INFO")

        try:
            response = requests.get(dataurl, timeout=15)
        except requests.exceptions.ConnectionError as errc:
            self.log(
[... 128 lines cut ...]
                        log="solar_log",
                        level="INFO",
                    )
            if entity == "solar_power_current":
                attributes = {**attributes, "update_minutes_ago": self.last_update}
                state = self.solar_power_current_state
            if entity == "solar_power_current_kw":
                state = self.solar_power_current_kw_state
            if entity == "solar_power_today":
                state = self.solar_power_today_state
            if entity == "solar_power_total":
                state = self.solar_power_total_state
            if entity == "solar_wifi_strength":
                attributes = {
                    **attributes,
                    "ip_address": self.ip_address,
                    "mac_address": self.mac_address,
                }
                state = self.solar_wifi_strength_state
            entity = f"sensor.{entity}"
            self.set_state(entity=entity, state=state, attributes=attributes)

        self.log("Solar states updated", log="solar_log")
        self.log("---------------------", log="solar_log")
```

robbinjanssen, 2021-08-30:
Closing in favor of #40

closed 2021-08-30 by robbinjanssen