robbinjanssen/home-assistant-omnik-inverter #29: error while omnik off

ELANDJEA, 2021-01-21:
my omnik shutsdown when there is not enough light anymore. The script starts throwing following error message 3 in a row every couple of minutes:

2021-01-21 19:32:27 ERROR (SyncWorker_9) [custom_components.omnik_inverter.sensor] Unable to fetch data from Omnik Inverter 192.168.x. x
2021-01-21 19:32:27 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.solar_power_total fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 278, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 474, in async_device_update
    raise exc
  File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/omnik_inverter/sensor.py", line 215, in update
    self._data.update()
  File "/usr/src/homeassistant/homeassistant/util/__init__.py", line 239, in wrapper
    result = method(*args, **kwargs)
  File "/config/custom_components/omnik_inverter/sensor.py", line 104, in _update
    fp.close()
UnboundLocalError: local variable 'fp' referenced before assignment

From my no coder knowlege, I think some error handling can be improved to a state when there is no data fetched to parse (for instance) value 0 back to HA.

I also notice that when HA is restarted during the omnik being shutdown, the sensors are not created and starts reporting "not available" in lovelace cards......by adding above error catching likely it will also keep creating the sensors during nightly restart of HA.

robbinjanssen, 2021-01-24:
Hi, I see what you mean. I'm not sure I can fix all of your problems, but I might be able to fix one. I'll try to create a new version this week!

robbinjanssen, 2021-02-05:
Hi, sorry I don't really have the time for this atm, i'll keep the issue open though!

ew-ewim, 2021-03-21:
Hey, interested as well in the solution of this issues as the add-on is hardly usable.  A pity as it hat a lot of potential !

robbinjanssen, 2021-04-02:
Hi @ew-ewim, I'm sorry to hear that. But fixing this would involve a (partly) rewrite of the component. One of the challenges is that the error "Unable to fetch data from Omnik Inverter 192.168.x.x" can have many reasons:
1. Is the IP address correct? 
2. Is the WIFI network down? 
3. Is the omnik (shut)down?

The code doesn't know the difference between 1, 2 and 3 and the cause of the error message is always the same: It simply can't reach the inverter. 

Upon setting up the component (which happens on each restart of you HA instance), it waits for data to be fetched and _then_ creates the HA sensors. This is because sensors need to have a unique ID and it is based on information received from your omnik inverter. If we want to tackle the problem above, we need to define sensors before-hand instead of waiting for response from the inverter. And for that I simply don't have the time at the moment. The code is pretty straight forward so you could try (re)writing it yourself and create a pull request. I'd be happy to review it and merge it :-)

robbinjanssen, 2021-08-30:
I think this is solved in v2 :-) See #40

closed 2021-08-30 by robbinjanssen