robbinjanssen/home-assistant-omnik-inverter #33: Not loading Entities

kevindebosscher, 2021-04-11:
Hello,

I"ve added via HACS, afterwards added below in the configuration.
```
sensor:
    - platform: omnik_inverter
      host: 192.168.1.114
```
Omnik folder is present in custom_components,how ever no Omnik solar entities showing up.

Best Regards

robbinjanssen, 2021-04-15:
hi @kevindebosscher, can you answer the following questions:
- What inverter do you have?
- Did you restart HA after adding the component via HACS?
- Do you have any errors in your logs?
- When you go to http://192.168.1.114/js/status.js or http://192.168.1.114/status.json?CMD=inv_query&rand=0.1234567 do you get any data?

kevindebosscher, 2021-04-16:
Hello Robbin,

-Omnik 4k
-Restart done like any other integration
-No errors but no omnik entity neither
-Pages dont open/work propably a wifi card than that has no json support?

robbinjanssen, 2021-04-26:
Hi @kevindebosscher

Might be a stupid question, but are you sure your inverter is connected to the Wifi?
If none of those pages work, then I'm afraid this integration won't work for you.

closed 2021-05-10 by robbinjanssen