robbinjanssen/home-assistant-omnik-inverter #36: Weird behavior when reading from two inverters

hepfister, 2021-05-08:
I have two inverters that I read using:
```
# Omnik Inverter 3k
  - platform: omnik_inverter
    host: 192.168.1.248
# Omnik Inverter 1.5K
  - platform: omnik_inverter
    host: 192.168.1.249
```
HA creates unique ID's, for example 'sensor.solar_power_today' and 'sensor.solar_power_today_2'. Yet, the data showed for 'sensor.solar_power_today_2' is the same as 'sensor.solar_power_today'. I know for a fact this amount should be very different.

The odd thing is that it only happens with 'sensor.solar_power_today'. 

'sensor.solar_power_total' and 'sensor.solar_power_today' are showing the correct values per inverters. Perhaps this is just luck.

I read in a different issue that the ID's are not pre-defined, but created dynamically when the inverter becomes online. Could this cause this issue when reading from >1 inverters with this plugin?

robbinjanssen, 2021-05-10:
Hi @hepfister, I think what you are describing is spot on. Could you try adding a 'name' (thanks to @Rmvandiepen in https://github.com/robbinjanssen/home-assistant-omnik-inverter/pull/35) to your config for each inverter? This will used when defining the entity names, that might resolve your issue!

closed 2021-05-18 by robbinjanssen

Rmvandiepen, 2021-05-19:
Found one small problem with the cache when using multiple inverters (the today_cache is shared between both inverters). Should be resolved by #37

robbinjanssen, 2021-05-21:
Awesome @Rmvandiepen! I've tagged 1.5.4 :)

hepfister, 2021-05-31:
@robbinjanssen, sorry for responding late. The 'name' suggestion works well. Testing your plugin alongside the other Omnik integration (which has problems on its own). Hope to replace it with your lighter weight implementation. Thanks for fixing the issue :)