robbinjanssen/home-assistant-omnik-inverter #38: omnik_inverter no longer supported in latest "core-2021.6.3"?

InjunBob, 2021-06-11:
Since upgrading to the latest core (absolutely NO other changes), the omnik_inverter custom integration seems no longer supported? Check config file results in "Platform error sensor.omnik_inverter - Integration 'omnik_inverter' not found.". Can anyone confirm / deny? Much appreciated! - Bob

robbinjanssen, 2021-06-24:
Hi @InjunBob 

Are you still having issues? I'm not having any issues at all and i'm on the latest version(s).

HA core-2021.6.6
HA: supervisor-2021.06.3
HACS: 1.13.2
Omnik Inverter: 1.5.4

InjunBob, 2021-06-24:
Hi Robbin,

Thanks for your intrest in my issue! At the time, I went back to the core 2021.5 version, which restored the functionality of the add-on (before I did that, I added a “version” statement to the manifest.json file, which I read was obligatory as of 2021.6, but that did not make a difference). Since you indicate you are now on 2021.6.6 (I tested on 2021.6.3), I will upgrade to that and see if I still have the problem - stay tuned.

Bob

InjunBob, 2021-06-24:
P.S. - I noticed that I am still on 1.5.0 version of Omnik, so I’ll upgrade first to the 1.5.4 version, to be in sync with you.

Bob

InjunBob, 2021-06-24:
Everything works, after upgrading both Omnik to 1.5.4 and Core to 2021.6.6 !

 

So not sure which one of the upgrades did the trick, but at least a big Thank You for the effort you put in Omnik Inverter.

 

Best regards,

 

Bob

robbinjanssen, 2021-06-25:
That's great! Closing this issue :-)

closed 2021-06-25 by robbinjanssen