robbinjanssen/home-assistant-omnik-inverter #39: No devices in status.js when no power is generated

freddieleeman, 2021-07-27:
When my inverter (PowerOne + iGEN Wi-Fi external logger) does not generate any power for a while, the inverter disappears from the status.js source, causing errors and resulting in 'Unavailable' sensors. 

_status.js with sun:_
```
var yz_device_num="1";
var myDeviceArray=new Array();
myDeviceArray[0]="xxxxxx          ,xxxx,,-xxxx-,,xxx,x,xxxxxx,,x,";
;
function initPageText()
```

_status.js without sun:_
```
var yz_device_num="0";
var myDeviceArray=new Array();
;
function initPageText()
```

**Possible fix:**
This issue could be resolved by using the cache when the source is retrievable (status.js found), but the regex for the data fails (or `yz_device_num` value is 0). The `power_current` sensor could be forced to 0, and the `power_today` and `power_total` values could be retrieved from cache.

klaasnicolaas, 2021-09-06:
Is there any way we can help you with this issue? It has been open for a while and in the meantime a lot of work has been done on a v2.0 version of this custom component.

robbinjanssen, 2021-09-07:
Before we can continue we need to make a decision on what to do with unreachable inverters. Feel free to jump in on https://github.com/klaasnicolaas/python-omnikinverter/discussions/24 :-)

robbinjanssen, 2021-09-22:
Created a new issue with more details about this :-)

closed 2021-09-22 by robbinjanssen