robbinjanssen/home-assistant-omnik-inverter #40: Home assistant's Energy Dashboard

mobiledude, 2021-07-30:
Latest Beta of HA contains an energy Dashboard. Would love to be able to use this integration to get the Energy out of the omnik invertor. However the entities must have specific characteristics to support this. Can you please have a look if this is something that can be supported? Would be great.  https://rc.home-assistant.io/blog/2021/07/28/release-20218/

freddieleeman, 2021-08-04:
For now, if you want to add `sensor.solar_power_total` to the Home Assistant Energy dashboard, just add this to your configuration.yaml:

```
homeassistant:
  customize_glob:
    sensor.solar_power_total:
      device_class: energy
      state_class: measurement
      last_reset: "1970-01-01T00:00:00+00:00"
```

wes1993, 2021-08-05:
Hello,
This should be wonderful but can we also add the power_current and power_today?
I'm have this in my HA config but can't see the sensor some suggestion?:

```
homeassistant:
  customize_glob:
    sensor.solar_power_total:
      device_class: energy
      state_class: measurement
      last_reset: "1970-01-01T00:00:00+00:00"
```

I can see the last reset value in my developer tools page

Best Regards
Stefano

robbinjanssen, 2021-08-06:
This looks awesome! As always PR's are welcome. Sorry, I don't have much spare time at the moment 👶 🎉

hepfister, 2021-08-06:
> For now, if you want to add `sensor.solar_power_total` to the Home Assistant Energy dashboard, just add this to your configuration.yaml:
>
> ```
> homeassistant:
>   customize_glob: [...]

Works just perfectly, many thanks!

freddieleeman, 2021-08-07:
> 
> 
> Hello,
> This should be wonderful but can we also add the power_current and power_today?
> 
> Best Regards
> Stefano

You cannot use those sensors for the Energy Dashboard. You could use 'power_today', but the last_reset value cannot be set static. I do not recommend doing this, and just use the power_total sensor for solar measurement.

wes1993, 2021-08-07:
@freddieleeman 
Thanks a lot for your reply :-D
Ok i will use this sensor and i have another question but i'm not sure if is possible, could we calculate the return to grid energy by subtracting current - shelly power sensor?

Best Regards
Stefano

klaasnicolaas, 2021-08-13:
> For now, if you want to add `sensor.solar_power_total` to the Home Assistant Energy dashboard, just add this to your configuration.yaml:
>
> ```
> homeassistant:
>   customize_glob: [...]

This approach is strongly discouraged, yes it allows an entity to appear in the energy dashboard. But in the long run can cause problems for the calculations in the long term statistics, depending on whether or not a device resets its values after a power cut etc.

In addition, from 2021.9 something will also change with the `last_reset` so you keep messing with it in the end 😉

hepfister, 2021-08-13:
> This approach is strongly discouraged, yes it allows an entity to appear in the energy dashboard. But in the long run can cause problems for the calculations in the long term statistics, depending [...]

What would you recommend doing instead? The sensor.solar_power_total is just adding up. This value is never reset regardless of power cuts (which happens every night). I don't see the long term problem. Can you explain a bit more?

freddieleeman, 2021-08-13:
the sensor.solar_power_total is a lifetime sensor that has been increasing over 6 years in my case. I don't see an issue with adding these three arguments to the sensor. It has been working fine with the Energy Dashboard. Of course it would be nicer if this integration would add the arguments automatically, but until then, I think the suggested customization is fine.

klaasnicolaas, 2021-08-13:
I mainly talk in general, in various places you see these suggestions passing by and people take them blindly and use them in their config, for one device it can work fine and for another it causes problems and people start complaining that it is not working properly.

But enough about it, I'll see if we can update the code for this integration, but I don't use it myself so could use some help with testing.

pixsoftware, 2021-08-19:
Could you add goodwe to the names, as it's not intuitive to fint the right sensor if you have many sensors in the system. maybe even add the plant name to the friendly name?

robbinjanssen, 2021-08-29:
So @klaasnicolaas has been working on an update and we are currently testing this. Looks like a v2 is coming soon with support for the energy dashboard :-)

robbinjanssen, 2021-08-30:
To continue with this we need people who use the `use_json` option. Please check out https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/41 if you can help :-)

robbinjanssen, 2021-09-14:
A pre-release has been tagged. You can update it using HACS if you have `Show Beta Releases` enabled.
https://github.com/robbinjanssen/home-assistant-omnik-inverter/releases/tag/v2.0.0-beta

## BREAKING
> **_This component can no longer be configured using `configuration.yaml`. You should remove it from your `configuration.yaml` and re-add it using [Home Assistant integrations](https://my.home-assistant.io/redirect/integrations/)_**

robbinjanssen, 2021-09-15:
Will be fixed in next release :)

closed 2021-09-15 by robbinjanssen