robbinjanssen/home-assistant-omnik-inverter #41: w

robbinjanssen, 2021-08-30:


robbinjanssen, 2021-08-30:
@GoSpursGoNL has the following inverter/data

**Type:** `omnik2000tl2`
**Firmware:** `VER:ME-111001-V1.0.6(2015-10-16)`
**JSON output**
```json
{
"g_sn":"x",
"g_ver":"VER:ME-111001-V1.0.6(2015-10-16)",
"time":1588598399,
"g_time":8002,
"g_err":0,
"mac":"xxx",
"auto_ip":1,
"ip":"192.168.1.28",
"sub":"255.255.255.0",
"gate":"192.168.1.9",
"auto_dns":1,
"dns":"8.8.8.8",
"dns_bak":"8.8.4.4",
"i_status":16,
"i_type":258,
"i_num":1,
"i_sn":"xxx",
"i_ver_m":"V1.25Build23261",
"i_ver_s":"V1.40Build52927",
"i_modle":"omnik2000tl2",
"i_pow":"2000",
"i_pow_n":1225,
"i_eday":"10.90",
"i_eall":"8674.0",
"i_alarm":"F23",
"i_last_t":0,
"a_agr":0,
"l_a_ip":"47.88.8.200",
"l_a_port":10000
}
```

relout, 2021-08-30:
{"g_sn":1905704036,"g_ver":"ME-121001-V1.0.6(201704012008)","time":31759255,"g_time":60,"g_err":"F02","mac":"xx.xx.xx.xx","auto_ip":1,"ip":"192.168.11.52","sub":"255.255.255.0","gate":"192.168.11.1","auto_dns":1,"dns":"192.168.11.1","dns_bak":"208.67.222.222","i_status":16,"i_type":1283,"i_num":1,"i_sn":"160E311910600420","i_ver_m":"","i_ver_s":"","i_modle":"","i_pow":"","i_pow_n":823,"i_eday":"1.70","i_eall":"2561.0","i_alarm":"F22F23","i_last_t":0,"l_a_agr":0,"l_a_ip":"115.29.186.234","l_a_port":10000,"agr_b":2,"ipdm_b":"","ip_b":"192.168.11.39","port_b":9999}

klaasnicolaas, 2021-08-30:
@relout Which model do you have?

relout, 2021-08-30:
Hoi Klaas,

De omvormer is van Solis, de mini 2000 4g.
En de inverter is van Ginlong.
Heb je hier voldoende aan?

Met vriendelijke groet,

René Elout

klaasnicolaas, 2021-08-30:
@relout Could we be in touch somehow (Slack or Discord) to run some tests? I have some code ready that could theoretically work with your inverter.

relout, 2021-08-31:
@klaasnicolaas i have send you an email with some possible ways to contact.
René

wes1993, 2021-09-03:
Hello,
Thanks a lot for yours wonderful work, unfortunately i don't have the Json inverter.
Just for information have you seen this?
https://community.home-assistant.io/t/how-can-i-add-total-increasing-to-my-template-sensor/334961/3

klaasnicolaas, 2021-09-03:
@wes1993 as an insider, all information in that community blog is already known to me 😉 

Optionally, you can also share your output with us from an inverter that retrieves data from the javascript file (Inverter IP/js/status.js). We have almost finished the integration and I expect that a first beta release will be coming soon.

wes1993, 2021-09-03:
@klaasnicolaas 
Thanks a lot for your reply!! This is wonderful!!! :)

wes1993, 2021-09-03:
This is the output of my JS Inverter (I think that if is possibile you could add also the RSSI sensor in HA):

var version= "H4.01.51MW.2.01W1.0.64(2018-01-251-D)";
var m2mMid = "HiddenByME";
var wlanMac= "HiddenByME";
var m2mRssi= "39%";
var wanIp= "192.168.0.34";
var nmac= "HiddenByME";
var fephy= "off";
var yz_device_type="258";
var yz_device_num="1";
var myDeviceArray=new Array();
myDeviceArray[0]="IELN1520178F5210,V4.08Build215,V4.12Build246,Omnik1500tl ,1500,850,232,52002,,1,";
;
function initPageText()
{
    var list = window.parent.reList("status");
    fileText("st1",list["t1"]);
    fileText("st2",list["t2"]);
    fileText("st3",list["t3"]);
    fileText("st4",list["t4"]);
    for(var i=1;i<=32;i++)
    {
        if(i!=14)
        {
            fileText("tx"+i,list[i]);
        }
    }
    init_main_page();
    initStatus();
    initInvList(parseInt(yz_device_type),yz_device_num);
    changeFont();
    child_getH();
}
function upfold(v)
{
    if(document.getElementById("up_"+v+"_div").style.display == "none")
    {
        show("up_"+v+"_div");
        document.getElementById("up_"+v).src = "image/up.png";
    }
    else
    {
        hide("up_"+v+"_div");
        document.getElementById("up_"+v).src = "image/down.png";
    }
    if(v == 3)
    {
        if(document.getElementById("ser_status").src=="")
        {
			//alert("mhtest");
            document.getElementById("ser_status").src="server_status.html"; 
        }
    }
}
function initInvList(type,num)
{
    if(type==null || type == "")
    {
        type = window.parent.reTip(40);
    }
    else
    {
        type = reInvBrand(type);
    }
    if(num==null || num == "")
    {
        hide("inv_msg_div");
        num = window.parent.reTip(40);
    }
[... 192 lines cut ...]
		    if(value*1==4294967295)
		    {
		        elem.innerHTML = "---";
		    }
		    else
		    {
		        sv= value*1/10;
			    elem.innerHTML= sv.toString()+" kWh";
		    }
		}
		else if (nv== 10)
		{
			if (document.getElementById("webdata_sn").innerHTML== "---")
				elem.innerHTML = "---";
			else
				elem.innerHTML = value+window.parent.reTip("5");
		}
		else
			elem.innerHTML = value;
	}
	else
	{
		alert(webdata_ids[nv]);
	}
}

wes1993, 2021-09-03:
Because some times I have some error and I think it’s related to the RSsI, if I possible just for tracking the connection and check if Wi-Fi it’s strong

Hjalmgunnarson, 2021-09-03:
This may be redundant, but here is the output of my Omniksol 3k TL2 (JS). Thank you for all the hard work!

`
var version="H4.01.38Y1.0.09W1.0.08";var m2mMid="0000000000";var wlanMac="00:00:00:00:00:00";var m2mRssi="0%";var wanIp="000.000.0.0";var nmac="000000000000";var fephy="off";var webData="NLBN300006C62000,NL2-V9.7-5930,V5.2-90155,omnik3000tld,3000,403,1432,162837,,4,";function initPageText(){var A=window.parent.reList("status");fileText("st1",A["t1"]);fileText("st2",A["t2"]);fileText("st3",A["t3"]);for(var B=1;B<=27;B++){if(B!=14){fileText("tx"+B,A[B])}}init_main_page();initStatus();changeFont();child_getH()}function upfold(A){if(document.getElementById("up_"+A+"_div").style.display=="none"){show("up_"+A+"_div");document.getElementById("up_"+A).src="image/up.png"}else{hide("up_"+A+"_div");document.getElementById("up_"+A).src="image/down.png"}if(A==3){if(document.getElementById("ser_status").src==""){document.getElementById("ser_status").src="server_status.html"}}}function init_main_page(){var E=document.form_cover_setting;cfg2Form(E);var C;var D;var B;var A;B=window.parent.reTip("1");A=window.parent.reTip("2");document.getElementById("cover_ver").innerHTML=version;document.getElementById("cover_mid").innerHTML=m2mMid;if(E.wifi_on.value=="disable"){document.getElementById("cover_ap_status").innerHTML=A;document.getElementById("cover_ap_mac").innerHTML="";document.getElementById("cover_ap_ssid").innerHTML="";document.getElementById("cover_ap_ip").innerHTML="";document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}else{if(E.sta_enable.value=="1"){document.getElementById("cover_ap_status").innerHTML=B;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=B;document.getElementById("cover_sta_rssi").innerHTML=m2mRssi;document.getElementById("cover_sta_ip").innerHTML=wanIp;C=wlanMac.substring(0,16);D="0x"+wlanMac.charAt(16);D=D*1+1;D=D.toString(16);D=D.toUpperCase();C=C+D;document.getElementById("cover_sta_mac").innerHTML=C;document.getElementById("cover_sta_ssid").innerHTML=E.sta_ssid.value}else{document.getElementById("cover_ap_status").innerHTML=B;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}}}function init_webdata_form(){var E=webData;var B="";var A,C,D;A=E.length;C=0;D=0;while(C<A){if(E.charAt(C)==","){C+=1;D+=1;web_data_show(D,B);B=""}else{B=B+E.charAt(C);C+=1}}}function web_data_show(D,A){var C=new Array("webdata_sn","webdata_msvn","webdata_ssvn","webdata_pv_type","webdata_rate_p","webdata_now_p","webdata_today_e","webdata_total_e","webdata_alarm","webdata_utime");if(D>10||D<=0){return}var B=document.getElementById(C[D-1]);if(A==""&&B!=null){B.innerHTML="---";return}if(B!=null){if(D==5){B.innerHTML=A+" W"}else{if(D==6){B.innerHTML=A+" W"}else{if(D==7){sv=A*1/100;B.innerHTML=sv.toString()+" kWh"}else{if(D==8){sv=A*1/10;B.innerHTML=sv.toString()+" kWh"}else{if(D==10){if(document.getElementById("webdata_sn").innerHTML=="---"){B.innerHTML="---"}else{B.innerHTML=A+window.parent.reTip("5")}}else{B.innerHTML=A}}}}}}else{alert(C[D])}};`

klaasnicolaas, 2021-09-03:
@Hjalmgunnarson Thnx for the output! Always welcome.

klaasnicolaas, 2021-09-03:
@wes1993 I'm now working on adding this, as well as IP and firmware version of your device.

wes1993, 2021-09-03:
@klaasnicolaas thanks a lot!! This is awesome!!!

GoSpursGoNL, 2021-09-04:
I wasn't using this integration anymore after I had stability issues with HA and decided to use as few as possible integrations. That is now all resolved. I had been using Node-Red to fetch the total overall consumption from the inverter and put it into a HA input_number entity.
But that isn't compatible with the Energy dashboard. With 2021.8 I could do a trick by converting it into a sensor via template sensor and then adding the energy device class
```
customize_glob:
    sensor.zonnepanelen_meterstand:
      device_class: energy
      state_class: measurement
      last_reset: "1970-01-01T00:00:00+00:00"
```
With 2021.9 that trick doesn't work anymore, as the energy settings page complains about the entity not having the 'expected state class 'total_increasing''. So perhaps now is a good moment for me to switch back to this integration.
Can I help with testing? You already posted my json output above.

robbinjanssen, 2021-09-06:
@GoSpursGoNL we'll release the beta soon, and we can use some testers then :-)

wes1993, 2021-09-06:
I can be a tester, if it’s ok for you.
Best regards
Stefano

robbinjanssen, 2021-09-07:
Before we can continue we need to make a decision on what to do with unreachable inverters. Feel free to jump in on https://github.com/klaasnicolaas/python-omnikinverter/discussions/24 :-)

robbinjanssen, 2021-09-14:
A pre-release has been tagged. You can update it using HACS if you have `Show Beta Releases` enabled.
https://github.com/robbinjanssen/home-assistant-omnik-inverter/releases/tag/v2.0.0-beta

## BREAKING
> **_This component can no longer be configured using `configuration.yaml`. You should remove it from your `configuration.yaml` and re-add it using [Home Assistant integrations](https://my.home-assistant.io/redirect/integrations/)_**

wes1993, 2021-09-14:
Hello,
I'm trying to install this but i can't find how to enable the beta release.
Could you help me
Best regards
Stefano

wes1993, 2021-09-14:
Hello, 
I have installed the Beta version:
[image]

But i can't see the integration in the integration page of HA:

[image]

[image]

I will wait your reply
Best regards
Stefano

klaasnicolaas, 2021-09-14:
@wes1993 Did you restart HA after installing?

wes1993, 2021-09-14:
Yes,
Also the entire server.
I don’t need to put anything in configuration.yaml is correct?

klaasnicolaas, 2021-09-14:
No more YAML code is needed 😄 I just installed it on my own HA environment and I can set up the integrations with the config_flow without any problems 🤔

wes1993, 2021-09-14:
Mmm I don’t know why… there is a way to see some log or other stuff?

robbinjanssen, 2021-09-14:
@wes1993 did you remove the configuration from your `yaml` file? Anyone else having the same problem?
[image]

wes1993, 2021-09-14:
@robbinjanssen 
Yes i have commented out the old config:
[image]

I'm trying to remove addon and reinstall

wes1993, 2021-09-14:
Yeeeeessss
Thanks a lot for everyone reinstalling and installing again (2 times) seems that have solved the problem :-D
I'll keep updated this thread if i found bugs etc :-D

Thanks again for your wonderful work :-D

wes1993, 2021-09-14:
There is a way to change the scan interval?

klaasnicolaas, 2021-09-14:
At the moment the scan_interval is 4 minutes, you can only change that by adjusting the python code. In the future we can optionally make this in the options config_flow.

wes1993, 2021-09-14:
Ok perfect!! This is wonderful :)

Hjalmgunnarson, 2021-09-14:
> A pre-release has been tagged. You can update it using HACS if you have `Show Beta Releases` enabled.
> https://github.com/robbinjanssen/home-assistant-omnik-inverter/releases/tag/v2.0.0-beta
>
> ## [...]

I'm having trouble installing the beta. It is not listed when I select "Install this repository in HACS". After that I select "Show Beta releases", but it is not listed. I restarted HASS once.

robbinjanssen, 2021-09-14:
@Hjalmgunnarson it looks like you need to install v1.5.4 first via HACS, after installing the v2.0.0-beta should be visible as an installation option.

wes1993, 2021-09-14:
@robbinjanssen 
Sorry again for the disturb, where is the line related to the scan interval? what file?

Thanks a lot
Stefano

robbinjanssen, 2021-09-14:
@wes1993 you can change it in [`const.py`](https://github.com/robbinjanssen/home-assistant-omnik-inverter/blob/dev/custom_components/omnik_inverter/const.py#L10). Note that when we update the component these changes are overiden.

wes1993, 2021-09-14:
@robbinjanssen 
Thanks a lot :-D
I know i hope that when an update was released i have this in the config, so for now it's ok :-D

Best Regards
Stefano

Hjalmgunnarson, 2021-09-14:
> @Hjalmgunnarson it looks like you need to install v1.5.4 first via HACS, after installing the v2.0.0-beta should be visible as an installation option.

Thanks, I thought I had to uninstall and reinstall, don't know why. It works now! Thanks a lot for all the work. I'll report back in a few days.

Hjalmgunnarson, 2021-09-15:
After approx 24 hours no apparent problems and no entries in home-assistant.log.

BobvanderWeg, 2021-09-16:
I encounter an issue with registering the inverter as a new repository. any hint?

[image]

klaasnicolaas, 2021-09-16:
Does your inverter transmit its data via a javascript file or JSON?

https://github.com/robbinjanssen/home-assistant-omnik-inverter/tree/dev#how-does-it-work

BobvanderWeg, 2021-09-16:
I tried both.
I have this output:

[image]

klaasnicolaas, 2021-09-16:
In that case don't check the "Gebruik JSON" box 😉

BobvanderWeg, 2021-09-16:
I tried without the "JSON" option also without result. It's a Hosola inverter but with version 1.5.2 it was working before.

klaasnicolaas, 2021-09-16:
Could you post the code in that file here? Then I'll test it locally.

BobvanderWeg, 2021-09-16:
var version="H4.01.51Y4.0.02W1.0.57(2017-08-311-D)";var m2mMid="640536678";var wlanMac="F0:FE:6B:C4:BC:C8";var m2mRssi="81%";var wanIp="192.168.178.22";var nmac="F0FE6BC4BCCB";var fephy="off";var yz_device_type="5633";var yz_device_num="1";var myDeviceArray=new Array();myDeviceArray[0]="PVL3000N18A21027,,,,,610,887,77304,,0,";
;function initPageText(){var B=window.parent.reList("status");fileText("st1",B["t1"]);fileText("st2",B["t2"]);fileText("st3",B["t3"]);fileText("st4",B["t4"]);for(var A=1;A<=32;A++){if(A!=14){fileText("tx"+A,B[A])}}init_main_page();initStatus();initInvList(parseInt(yz_device_type),yz_device_num);changeFont();child_getH()}function upfold(A){if(document.getElementById("up_"+A+"_div").style.display=="none"){show("up_"+A+"_div");document.getElementById("up_"+A).src="image/up.png"}else{hide("up_"+A+"_div");document.getElementById("up_"+A).src="image/down.png"}if(A==3){if(document.getElementById("ser_status").src==""){document.getElementById("ser_status").src="server_status.html"}}}function initInvList(A,C){if(A==null||A==""){A=window.parent.reTip(40)}else{A=reInvBrand(A)}if(C==null||C==""){hide("inv_msg_div");C=window.parent.reTip(40)}else{init_invdata_form()}var D=document.getElementById("inv_type");if(D){D.innerHTML=A}var B=document.getElementById("inv_num");if(B){B.innerHTML=C}}function init_main_page(){var E=document.form_cover_setting;cfg2Form(E);var C;var B;var D;var A;D=window.parent.reTip("1");A=window.parent.reTip("2");document.getElementById("cover_ver").innerHTML=version;document.getElementById("cover_mid").innerHTML=m2mMid;if(E.wifi_on.value=="disable"){document.getElementById("cover_ap_status").innerHTML=A;document.getElementById("cover_ap_mac").innerHTML="";document.getElementById("cover_ap_ssid").innerHTML="";document.getElementById("cover_ap_ip").innerHTML="";document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}else{if(E.sta_enable.value=="1"){document.getElementById("cover_ap_status").innerHTML=D;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=D;document.getElementById("cover_sta_rssi").innerHTML=m2mRssi;document.getElementById("cover_sta_ip").innerHTML=wanIp;C=wlanMac.substring(0,16);B="0x"+wlanMac.charAt(16);B=B*1+1;B=B.toString(16);B=B.toUpperCase();C=C+B;document.getElementById("cover_sta_mac").innerHTML=C;document.getElementById("cover_sta_ssid").innerHTML=E.sta_ssid.value}else{document.getElementById("cover_ap_status").innerHTML=D;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}}if(fephy=="on"){document.getElementById("cover_cable_status").innerHTML=D;document.getElementById("cover_cable_ip").innerHTML=wanIp;document.getElementById("cover_cable_mac").innerHTML=nmac.substring(0,2)+":"+nmac.substring(2,4)+":"+nmac.substring(4,6)+":"+nmac.substring(6,8)+":"+nmac.substring(8,10)+":"+nmac.substring(10,12)}else{document.getElementById("cover_cable_status").innerHTML=A;document.getElementById("cover_cable_ip").innerHTML="";document.getElementById("cover_cable_mac").innerHTML=""}}function init_invdata_form(){var E="";var D=document.getElementById("webdata_sn");D.innerHTML="";if(D&&myDeviceArray.length>0){show("inv_msg_div");var F=myDeviceArray[0].split(",");for(var A=0;A<myDeviceArray.length;A++){var B=document.createElement("option");var C=myDeviceArray[A].split(",");B.value=C[0];B.innerHTML=C[0];D.appendChild(B)}for(var A=1;A<F.length;A++){inv_data_show((A+1),F[A])}}}function change_inv_list(){var B=document.getElementById("webdata_sn");if(B){var C=B.selectedIndex;if(myDeviceArray.length>C){var D=myDeviceArray[C].split(",");for(var A=1;A<D.length;A++){inv_data_show((A+1),D[A])}}}}function inv_data_show(D,C){var A=new Array("webdata_sn","webdata_msvn","webdata_ssvn","webdata_pv_type","webdata_rate_p","webdata_now_p","webdata_today_e","webdata_total_e","webdata_alarm","webdata_utime");if(D>10||D<=0){return}var B=document.getElementById(A[D-1]);if(C==""&&B!=null){B.innerHTML="---";return}if(B!=null){if(D==5){B.innerHTML=C+" W"}else{if(D==6){if((C*1==4294967295)||(C*1 == 0)){B.innerHTML="---"}else{B.innerHTML=C+" W"}}else{if(D==7){if(C*1==4294967295){B.innerHTML="---"}else{sv=C*1/100;B.innerHTML=sv.toString()+" kWh"}}else{if(D==8){if(C*1==4294967295){B.innerHTML="---"}else{sv=C*1/10;B.innerHTML=sv.toString()+" kWh"}}else{if(D==10){if(document.getElementById("webdata_sn").innerHTML=="---"){B.innerHTML="---"}else{B.innerHTML=C+window.parent.reTip("5")}}else{B.innerHTML=C}}}}}}else{alert(A[D])}};

klaasnicolaas, 2021-09-16:
Thank you! I see what's going wrong and will fix it soon with a new beta release.

Hjalmgunnarson, 2021-09-18:
The signal quality reported is very erratic. Is anyone else experiencing this?
[image]

wes1993, 2021-09-18:
I have similar situation but it’s not a bug of the software but the shit FW and poor quality antenna of the Wi-Fi Module for Inverter

wes1993, 2021-09-18:
My situation:
[image]

klaasnicolaas, 2021-09-19:
If this is really an issue, please create a new one 😉

robbinjanssen, 2021-09-20:
@Hjalmgunnarson @wes1993 I think its indeed just a shit FW with a shitty antenna. Mine looks like this:
[image]

I think it's something we have to accept :)

wes1993, 2021-09-20:
Yes Infact I have asked to add the sensor for monitoring the WI-FI.
This is normal but moving a little bit the antenna I have strongest Wi-Fi also if drop connection

Hjalmgunnarson, 2021-09-20:
> @Hjalmgunnarson @wes1993 I think its indeed just a shit FW with a shitty antenna. Mine looks like this:
> [image]
> 
> I think it's something we have to accept :)

I can live with that. :hurtrealbad:

robbinjanssen, 2021-09-22:
Thanks to @klaasnicolaas [v2.0.0-beta-2](https://github.com/robbinjanssen/home-assistant-omnik-inverter/releases/tag/v2.0.0-beta-2) has been released 🎉

BobvanderWeg, 2021-09-22:
In my test environment it's working to connect to the inverter and read the values. thanks

NoWiGit, 2021-09-24:
Type: omnik2000tl
Firmware: NL1-V1.0-0043-4
JSON output: none

robbinjanssen, 2021-09-25:
> Type: omnik2000tl
> Firmware: NL1-V1.0-0043-4
> JSON output: none

And this is working for you? Or do you have `status.js` output? :)

NoWiGit, 2021-09-26:
I'm using _status.js_ I guess ;-)

Thanks for your great work on this custom component. beta2 is working like a charm.

wes1993, 2021-09-29:
Error in Beta-3

Error setting up entry Fotovoltaico for omnik_inverter
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 304, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/config/custom_components/omnik_inverter/__init__.py", line 37, in async_setup_entry
    coordinator = OmnikInverterDataUpdateCoordinator(hass, scan_interval)
  File "/config/custom_components/omnik_inverter/__init__.py", line 92, in __init__
    if self.config_entry.data[CONF_SOURCE_TYPE] == "html":
KeyError: 'source_type

[1 more reply]

closed 2021-10-03 by robbinjanssen