robbinjanssen/home-assistant-omnik-inverter #45: Status.html support?

Jackahack, 2021-09-19:
i use an omnik 2500tl. this router does not has a json or js page but it just posts it in a status.html. is it possible to support this? as it is direct webdata  is directly visible in the html code? any suggestions?

klaasnicolaas, 2021-09-19:
Are you sure it doesn't have a javascript file? We can make support for it, just don't think it will be in the short term.

Jackahack, 2021-09-19:
yeah it has only a javascript code block in the status.html: 
```
</script>
<script type="text/javascript">
var webdata_sn = "[redacted]";
var webdata_msvn = "V5.07Build245";
var webdata_ssvn = "V4.13Build262";
var webdata_pv_type = "Omnik2500tl ";
var webdata_rate_p = "2500";
var webdata_now_p = "81";
var webdata_today_e = "10.82";
var webdata_total_e = "6454.3";
var webdata_alarm = "";
var webdata_utime = "0";
var cover_mid = "[redacted[";
var cover_ver = "ME_08_0102_2.03";
var cover_wmode = "";
var cover_ap_ssid = "";
var cover_ap_ip = "";
var cover_ap_mac = "[redacted";
var cover_sta_ssid = "";
var cover_sta_rssi = "";
var cover_sta_ip = "192.168.0.106";
var cover_sta_mac = "[redacted];
var status_a = "1";
var status_b = "0";
var status_c = "0";

```

robbinjanssen, 2021-09-20:
@Jackahack can you paste the complete output of status.html?

Jackahack, 2021-09-20:
@robbinjanssen 
 the javascript code block starts at line 172
[status_html_output.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/7194223/status_html_output.txt)

klaasnicolaas, 2021-09-23:
What is the link to the file and is it accessible with or without login?

Jackahack, 2021-09-24:
@klaasnicolaas http://<ip-adress>/status.html 
i do have to login with a basic browser login window. nothing to fancy.

klaasnicolaas, 2021-09-24:
Well then I know enough, with a little luck support will already be in the next beta release.

klaasnicolaas, 2021-09-25:
@Jackahack In the meantime I'm working on adapting the integration so that it can handle your HTML type or inverter. Could you contact me privately (Email, Twitter, Discord etc.) so we can invite you to Slack, share some stuff so you can test a modified version soon 😉

Jackahack, 2021-09-26:
@klaasnicolaas yes sure.  Discord works good enough. Here is my tag: MegaJule#1536

klaasnicolaas, 2021-09-26:
I sent you a message on Discord 😉

closed 2021-09-27 by klaasnicolaas

Jackahack, 2021-12-21:
ok. it worked for me a short while and then it stopped working.  i kinda forgot about it thats why it took me so long. but now i am reporting it.  i get the following error:
 `[homeassistant.config_entries] Config entry 'omnik_pv' for omnik_inverter integration not ready yet: Error occurred while communicating with Omnik Inverter device; Retrying in background`. 

Nothing happens. it does not  create entities.  it does not start.  i tried uninstalling the hacs integration and the integration and then reinstalling. i even tried different versions.

PR #53 (2021-09-27): Add support for HTML sources
- ✨ Add support for HTML source inverters
- 📓 Update the strings of the integration
- 📚 Update the documentation

close #45