robbinjanssen/home-assistant-omnik-inverter #51: Add caching layer for inverters that go offline after dark

robbinjanssen, 2021-09-22:
We've learned that some inverters go offline after dark. Meaning the status.js/json file is no longer reachable. Instead of throwing errors and exceptions we would like to introduce a caching layer. As discussed in the [python client](https://github.com/klaasnicolaas/python-omnikinverter) for this package we need to add the layer to this integration. 

@sander76 reached out and pointed us in the right direction. Unfortunately I don't have much time right now, it would be awesome if anyone would like to give it a go!

Instructions here:
https://github.com/klaasnicolaas/python-omnikinverter/discussions/24#discussioncomment-1323129

klaasnicolaas, 2021-09-27:
Let's sign this up for hacktoberfest 😄

ronaldt80, 2021-10-05:
@klaasnicolaas is there any way to correct faulty values in the long-term statistics database?
I have 2 inverters, 1 went briefly offline and back online twice today. Result is that HA Energy showed 13,000kwh production today. The previous integration had caching on so never experienced this issue before. Will probably roll back to v1 until caching is possible in V2 as well but in the meantime would be good to correct the incorrect data.

klaasnicolaas, 2021-10-12:
As stated before, this has to do with your inverter sending incorrect values at certain times (we have seen this on JSON source). To determine this for sure, we need to apply some extra logging so that we can figure this out. The `last_reset` has nothing to do with it, it only assumes that there has been a reset due to a decreating in the `total_increasing`.

What source do you use and could you access it privately so we can debug this further?

ELANDJEA, 2021-10-13:
I used the JS source for getting the data. (http://<your omnik ip address>/js/status.js)
if needed I can get output from that source.....guess one time output is not enough....you need at least a certain sequence?

wes1993, 2021-11-19:
We have some news about this implementation?

robbinjanssen, 2021-11-25:
Unfortunately not, currently i dont have much spare time. But i'm happy to accept a PR! :)

freddieleeman, 2023-01-31:
I hope someone is still willing to look at this. I'm currently solving this by adding an SQL query that retrieves the latest non-"unavailable" value from the states. But it could be more convenient to have this feature in the integration.

robbinjanssen, 2023-01-31:
I've actually started working on this :-) but ran into some issues, more updates soon!

robbinjanssen, 2023-02-13:
Im having trouble implementing this as I cannot reproduce this with my own inverter.

Can someone tell me if their inverter becomes completely unavailable via the web as well? Or is it only the inverter and is the device still online?

freddieleeman, 2023-02-14:
day:
[image]

night:
[image]

js: https://pastebin.com/pvr3VPVA

timn, 2023-02-18:
The Bosswerk MI-600 goes completely dark, it never draws from the grid but only from solar power. Therefore it's offline when no sun is shining.

Would this caching layer also solve the problem that for a little while after the inverter booted up the total is showing zero before then shooting to the actual value? That's the problem that brought me here (backstory: my inverter has no internet access, because of that it does not reset its daily yield, using an utility meter with daily rotation on total yield should replace that, but it would probably produce invalid sums when there are 0 values and then totals again, I'll know for sure tomorrow, I just set this up).

timn, 2023-02-18:
I can already confirm a similar problem: using the solar_energy_total in the energy dashboard leads to a high colum at the very beginning with the accumulated total on every day (I thought this was due to the not resetting daily yield, but it's actually the total yield, so already the energy dashboard is broken by this, and not just the utility meters I wanted to set up). This is caused by the value being zero for the first few measurements every day and then having the actual value. I can filter this with a template sensor now, but it'd be great if the integration would already do that by itself (it should be safe to always reject 0 for the total yield, that will only exist for a very short time and it should be ok not to report that).

freddieleeman, 2023-04-15:
I fixed it with an SQL integration:

```
      SELECT
        states.state
      FROM
        states
        LEFT JOIN state_attributes ON (
          states.attributes_id = state_attributes.attributes_id
        )
      WHERE
        metadata_id = (
          SELECT
            metadata_id
          FROM
            states_meta
          where
            entity_id = 'sensor.powerone_solar_energy_total'
        )
      AND states.state > 1 AND states.state != 'unknown' and states.state != 'unavailable' 
      ORDER BY
        state_id DESC
      LIMIT
        1;
```

But this is far from convenient. It would be great if the integration would ignore unavailable/unknown/0.

PyroPath, 2024-01-04:
I've a rather old SolarBright 3.0kW for 12 panels on my main roof which also goes completely dark and APSystems DS3 for small cluster of XL panels on a little building in my garden. I have a kWh meter between my inverters and the grid to have a cumulative value for all solar (shame on you APSystems for reporting DC side yield stats!). This kWh meter also confirms there is 0 power draw after the sun goes down and the unit offline.

I have a tick box for using cached values, but this clears when leaving the configuration screen and thus doesn't work. It would be nice if we could get a solution for all the errors in the logbook.

Floriszz, 2024-12-06:
Today I had to change the IP of the invertor. Mine goes completely of after dark as well.
Has this option, "cache the total power of today" to do do with the requested feature?
[image]