robbinjanssen/home-assistant-omnik-inverter #56: Beta 3 Won’t Work

wes1993, 2021-09-29:
Hello, after I have updated to beta 3 I have this error in the log
```
Error setting up entry Fotovoltaico for omnik_inverter
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 304, in async_setup
    result = await component.async_setup_entry(hass, self)  # type: ignore
  File "/config/custom_components/omnik_inverter/__init__.py", line 37, in async_setup_entry
    coordinator = OmnikInverterDataUpdateCoordinator(hass, scan_interval)
  File "/config/custom_components/omnik_inverter/__init__.py", line 92, in __init__
    if self.config_entry.data[CONF_SOURCE_TYPE] == "html":
KeyError: 'source_type
```

klaasnicolaas, 2021-09-29:
Have you tried removing and adding the integration again?

wes1993, 2021-09-30:
I have tried but whitout any luck.

robbinjanssen, 2021-09-30:
@wes1993 just to clarify; did you remove and re-add the [integration](https://my.home-assistant.io/redirect/integrations/)?

wes1993, 2021-10-02:
Hello @robbinjanssen,
Sorry for the delay to reply :(
I have deleted and installed the RC version and I don’t know why but now works very well :)

Thanks a lot for the wonderful work :)

closed 2021-10-02 by wes1993