robbinjanssen/home-assistant-omnik-inverter #77: Ginlong stick password

kongjudas, 2021-11-01:
Get "Failed to connect". I have probably overseen something basic here, but when I try to access my Ginlong stick via the browser, I'm asked for user/ password. I can't see how the integration could get pass that with the GUI installation.

klaasnicolaas, 2021-11-01:
What do you see when you go to this page? Do you also have to log in? 
`http://<your omnik ip address>/status.json?CMD=inv_query`

kongjudas, 2021-11-01:
ERROR:404 Not Found

klaasnicolaas, 2021-11-01:
There is really no way to determine how those Ginlong sticks work :(, it works differently for everyone, making support for them extra difficult.

kongjudas, 2021-11-01:
Well, I've found that it serves the data thru HTML. So far so good, but no cigar.

robbinjanssen, 2021-11-10:
@kongjudas can you output the data that is served by the Inverter? Maybe that differs from what we expect?

closed 2021-11-24 by klaasnicolaas