robbinjanssen/home-assistant-omnik-inverter #81: Leave old value instead of Unavailable

wes1993, 2021-11-17:
Hello,
We have some problem with this inverter that goes in unavailable state very often, instead of go in unavailble also in HA sensors could we leave the latest value readed? 
Se we don't have all this sensors unavailable also in HA, a little explanation:
If inverter not reply:
solar_current_power = 0
solar_energy_today = last value before connection lost
solar_energy_total =  last value before connection lost
Best Regards
Stefano

robbinjanssen, 2021-11-18:
@wes1993 therefor we will need some sort of caching layer, there already is an issue for this. Closing this issue :)

Duplicate of https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/51

closed 2021-11-18 by robbinjanssen

wes1993, 2021-11-19:
Sorry @robbinjanssen 
Thanks a lot and i hope you could find a way to add this feature

Bye
Stefano