robbinjanssen/home-assistant-omnik-inverter #85: Error occurred while communicating with Omnik Inverter device; Retrying in background

Jackahack, 2021-12-21:
i own the 2500 TL; Status.html.

This integration did work for a time,  but then it wouldnt connect all of a sudden.  i got the following error: 
`[homeassistant.config_entries] Config entry 'omnik_pv' for omnik_inverter integration not ready yet: Error occurred while communicating with Omnik Inverter device; Retrying in background.

Nothing ever happens again except for the same error.  It does not start, it does not create entities. 

This may be something locally related or not.

robbinjanssen, 2021-12-22:
Hi @Jackahack,

This seems like a local issue.

Some inverters switch off automatically if no power is collected, can you try if you can access the status page at the same time you get the error?

Jackahack, 2021-12-22:
@robbinjanssen  so my inverter does generate power currently. but the integration keeps throwing the error. i can view the status.html in the browsers debugger. the data is there... 

although when i first opened my browsers debugger on the <ip>/status.html page. it would not load immedietly it also threw an error about fetching data and reloading the page. after that the debugger showed the data properly. but the integration stil has the error.

Jackahack, 2021-12-22:
to close out a potential local setup problem. i created a clean homeassistant os instance in a vm.  the integration still gives the error.

robbinjanssen, 2021-12-22:
@Jackahack i've just tried re-adding the integration myself and it still works.

So if I understand correctly adding the integration works, but after adding the devices/entities they no longer show up?

Jackahack, 2021-12-22:
@robbinjanssen  the adding of the integration works. but it does not read data and there fore it does not create entities/devices.  so yes thats whats happens.   the error states the integration is not ready because there is a communication error between HA and the inverter.

robbinjanssen, 2021-12-23:
That's strange, I think we need to add some more logging or invite you to our slack channel so we can debug directly :-) 
Could you tell us which version of HA and the integration you are using?

robbinjanssen, 2021-12-26:
We talked on slack and the issue seemed to have resolved itself! :-)

[image]

closed 2021-12-26 by robbinjanssen