robbinjanssen/home-assistant-omnik-inverter #99: Feature request: Add automated upload functionality to PVoutput.org

hepfister, 2022-02-16:
## Detailed description

## Context

klaasnicolaas, 2022-02-16:
It would be a good idea to see if we can upload that data to PVoutput. The functionality would then have to be created in the [python package](https://github.com/klaasnicolaas/python-omnikinverter) and the interval execution in the integration.

I'm just not that familiar with the platform myself and how easy it is to push that data to the platform.

robbinjanssen, 2022-02-16:
@hepfister I understand your problem, but I don't think its within the scope of this component. This is purely a "read data form inverter and sent to HA"-component. If we start supporting sites like PVoutput, who knows where it will end.

I think you should take a look at https://github.com/Woutrrr/Omnik-Data-Logger, it's a python script that does exactly what you want I think :)

closed 2022-02-16 by robbinjanssen