robbinjanssen/home-assistant-omnik-inverter #103: Error message when installing and setting up the integration

Wo0Zzz, 2022-03-15:
Hi there,

Just found out this solution to read my PV panels information in Home Assistant.
When setting up I ran into some problems.
The first was the "config flow could not be loaded".
Found out it was an MQTT issue, didn't install it yet. 
When i installed MQTT I get the set up page, when I add my to inverters it works great.
But when I wan't to add the entity's to my Energy dashboard, can not find them.
Searching a little bit further gives me this error message: 

"Logger: homeassistant.setup
Source: setup.py:317
First occurred: 23:22:28 (2 occurrences)
Last logged: 23:22:45

Unable to prepare setup for platform omnik_inverter.sensor: Platform not found (cannot import name 'SensorDeviceClass' from 'homeassistant.components.sensor' (/usr/src/homeassistant/homeassistant/components/sensor/__init__.py))."

What am I doing wrong? Still learning the magic of Home Assistant and programming, But when Googling this error, can't get any further. 

Thanks in advance!

klaasnicolaas, 2022-03-15:
MQTT is completely outside this integration, we don't use it at all.

Which Home Assistant version are you running?

Wo0Zzz, 2022-03-16:
Okay, I think that would be a coincidence then.
My HA is running on version 2021.9.6 because when I want to update the whole system freezes. 
Could that be the issue?

klaasnicolaas, 2022-03-16:
That explains everything, with Home Assistant it is important to keep up with updates and you are now simply missing crucial changes from core.

Wo0Zzz, 2022-03-16:
Hi there, 
Finally managed to update my HA instance to version 2022.3.5. 
When adding the Omnik Inverter integration, it keeps hanging on: "Please wait while Omnik Inverter is being set up"
See attached screenshot: 
[image]
It is in Dutch.
It doesn't go further, what am I doing wrong? 
I added the files from the Github to my "custom-components" folder, restarted HA and tried to add the Omnik Inverter entity, when I'm doing this I'll keep the screen as in the screenshot.
Hope you can help me to get to a solution!

Wo0Zzz, 2022-03-16:
Remembered what I did yesterday, as you said, you don't use MQTT, but I also installed the AppDaemon. 
That's my solution! Now the integration is working! Thank you! So the first step is to update you;re HA to the newest version and second step is to install the AppDaemon! Didn't know both of them! :-)

klaasnicolaas, 2022-03-16:
Also AppDeamon has nothing to do with this integration....

That the screen hangs is not entirely true, in the background HA is installing the corresponding python package. Just a matter of waiting after that you can continue with the integration.

robbinjanssen, 2022-03-17:
@Wo0Zzz seems like everything is working now, good! 👍

closed 2022-03-17 by robbinjanssen