robbinjanssen/home-assistant-omnik-inverter #104: No connection to Sofar inverter 3.6ktlm-g2

ploman1973, 2022-04-05:
I am trying to connect to my Sofar inverter with Wifi stick (serial 0528...). Al the the methods do not make a connection.

I can connect to my logger with 192.168.86.194/status.html and see this info in the source:

<html>
<body>

</style>
--
  | <script type="text/javascript">
  | var height=0;function fileText(id,value){if(document.getElementById(id)){document.getElementById(id).innerHTML=value}}function changeFont(){reCon("main_div").style.fontFamily=window.parent.reFont()}function child_getH(){var nh=document.body.offsetHeight+100;if(nh<500\|\|nh==null){nh=500}if(height!=nh){height=nh;window.parent.child_height(height)}}function reCon(id){return document.getElementById(id)}function ready(){try{window.parent.show_ifr()}catch(e){}child_getH()}function show(v){var c=document.getElementById(v);if(c!=null){c.style.display=""}}function hide(v){var c=document.getElementById(v);if(c!=null){c.style.display="none"}};
  | </script>
  | <script type="text/javascript">
  | var webdata_sn = "SH1ES136K3V033  ";
  | var webdata_msvn = "V140";
  | var webdata_ssvn = "";
  | var webdata_pv_type = "SH1ES136";
  | var webdata_rate_p = "";
  | var webdata_now_p = "110";
  | var webdata_today_e = "0.9";
  | var webdata_total_e = "3375.0";
  | var webdata_alarm = "";
  | var webdata_utime = "0";
  | var cover_mid = "5208......";
  | var cover_ver = "MW_08_2701_1.27";
  | var cover_wmode = "STA";
  | var cover_ap_ssid = "AP_520898588";
  | var cover_ap_ip = "10.10.100.254";
  | var cover_ap_mac = "98D8634BC497";
  | var cover_sta_ssid = "........";
  | var cover_sta_rssi = "78%";
  | var cover_sta_ip = "192.168.86.194";
  | var cover_sta_mac = "98D8634BC496";
  | var status_a = "1";
  | var status_b = "0";
  | var status_c = "0";

</body>
</html>

robbinjanssen, 2022-04-05:
Hi @ploman1973, can you paste the complete status.html between ` ```  ` please? Did you add the `|` yourself or are they there?

robbinjanssen, 2022-04-05:
Thanks, do you see any error whatsoever? Could it be that your HA has no access to that IP address?

ploman1973, 2022-04-05:
Hello, the only error is the message that the site cannot be found. (see picture, I am sure user and password are correct) Both logger en Homeassistant are in my local network.
[image]

robbinjanssen, 2022-04-05:
@ploman1973, try using only the IP address (`192.168.68.194`) without HTTP and status.html as host, that should work :)

ploman1973, 2022-04-05:
pfffff thought I had tried this already a 1000 times :). But not because this works. Thank you verry much

closed 2022-04-05 by ploman1973