robbinjanssen/home-assistant-omnik-inverter #113: A username/password is missing error after update to 2.4.0

mvdwetering, 2022-06-04:
## Detailed description

After updating from 2.3.0 to 2.4.0 the integration does not load anymore. It shows the following error in HA.

>Config entry 'Solis Omvormer' for omnik_inverter integration not ready yet: A username and/or password is missing from the request; Retrying in background

This is with a Ginlog Solis Wifistick that used to work before the update.

There are a username and password in the config entry and they are still valid (checked with a browser)

```
            {
                "entry_id": "30c76ef9eb2272bbf72832b70d4c700f",
                "version": 2,
                "domain": "omnik_inverter",
                "title": "Solis Omvormer",
                "data": {
                    "host": "192.168.178.32",
                    "source_type": "html",
                    "username": "**username**",
                    "password": "**password**"
                },
                "options": {},
                "pref_disable_new_entities": false,
                "pref_disable_polling": false,
                "source": "user",
                "unique_id": null,
                "disabled_by": null
            },
```

## Context

Not sure what to write here.

## Diagnostics

I don't see the diagnostics option for this integration?

MarijnS95, 2022-06-04:
This seems to be a:

Duplicate of #112

closed 2022-06-04 by klaasnicolaas

mvdwetering, 2022-06-04:
Thanks for the quick fix, works great 👍

MarijnS95, 2022-06-04:
My apologies for breaking it in the first place, that was a poor 2-letter typo!

PR #114 (2022-06-04): Prevent overwriting `omnikinverter` with authentication-less instance on "html" source type
Fixes #112, fixes #113

This should have been an `elif`, now causing `source_type = "html"` to fall into the `else` branch of the check for `"tcp"` and creating an `OmnikInverter` instance without username and password.