robbinjanssen/home-assistant-omnik-inverter #116: Unexpected error fetching omnik_inverter data: coroutine ignored GeneratorExit

serrj-sv, 2022-06-09:
I get this error every now and then since switching to TCP mode for my Omnik 3000TL:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 191, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/omnik_inverter/__init__.py", line 125, in _async_update_data
    SERVICE_INVERTER: await self.omnikinverter.inverter(),
  File "/usr/local/lib/python3.9/site-packages/omnikinverter/omnikinverter.py", line 186, in inverter
    fields = await self.tcp_request()
RuntimeError: coroutine ignored GeneratorExit
```

serrj-sv, 2022-06-09:
I was able to mitigate it by adding ```async with async_timeout.timeout(10):``` at line 126 as per official HA documentation:
https://developers.home-assistant.io/docs/integration_fetching_data/

MarijnS95, 2022-06-10:
I've also seen this `RuntimeError GeneratorExit` error a couple times, but don't yet understand where it comes from. I'd argue a timeout in the underlying socket would bubble up as `TimeoutError`, but that doesn't seem the case (or this is a different kind of error).

Perhaps this workaround makes the `async` backend timeout before that `RuntimeError` is raised; and as mentioned by that comment it seems HA gracefully handles a `TimeoutError` without spamming logs full.

@klaasnicolaas pointed out that this is also done for the other (JS/HTML) data sources:

https://github.com/klaasnicolaas/python-omnikinverter/blob/6173f341ad4266b1674a3fb6879e792c6cdd3c07/omnikinverter/omnikinverter.py#L89

We should probably quickly release a followup that adds this to the TCP backend, but I'd like to get the conversion from poll to push over with sooner rather than later (where this is handled differently or not at all).

Thanks for testing out the new TCP backend though! While I don't like it, it's technically still early days and bugs have to be fully ironed out :crossed_fingers:

MarijnS95, 2022-06-11:
This mitigation is now up in https://github.com/klaasnicolaas/python-omnikinverter/pull/207, still unsure where the error is coming from...

ianfretwell, 2023-02-03:
Is there a fix for this actually applied?  I'm running HA 2023.2 and still getting this same error (149 times since installing 2023.2 yesterday).

MarijnS95, 2023-02-03:
Nope.

ianfretwell, 2023-02-03:
Fair enough!

closed 2023-05-29 by klaasnicolaas