robbinjanssen/home-assistant-omnik-inverter #117: Can't add Omnik omnisol 1.5k-TL

Videodok, 2022-06-12:
I can't seem to add the omnisol1.5k-TL, none of the protocols seems to work, none of the test links for checking the status seem to work either. I'm submitting my data to solarman smart, so there is data to be found somewhere. Is my inverter of 2013, too old? 

## Detailed description

## Context

## Diagnostics

```json
"Copy here the code or upload the file"
```

MarijnS95, 2022-06-12:
@Videodok Are you sure your Omnik inverter is reachable from the machine that you are running HASS on? That is, if you `ping <omnik IP address>`, does it reply?

Videodok, 2022-06-12:
HASS is running on a Synology NAS, just SSH-ed into the synology to ping the Omnik inverter from there, without any problems. So yes, the inverter is reachable.

MarijnS95, 2022-06-12:
And if you run `curl http://<your omnik ip address>/js/status.js` from there?

Videodok, 2022-06-13:
I get a 404 error. So I don't get an js/HTML page whatsoever as described. I have seen only the initial config page and the fact that I get stats from solarman smart, insures me there is data transfer...

Videodok, 2022-06-13:
> :
> Do you have any idea how the data is submitted to solarman? :)

Honestly I don't. After the bankruptcy of Omnik my details were transferred
from Omnik to Solarman. I know you need to insert the serial of the Unit
and the serial and of the logger/wifi module. But I can't configure
anything regarding sending the data on the logger/wifi module. I only have
1 configuration page, where I can setup the wifi connection, keys and so
on. That's it.

robbinjanssen, 2022-06-15:
Are you sure you are entering the correct serial? This can be confusing. You need to have the one from the (wifi) device information.

[image]

Videodok, 2022-06-15:
yes, the only option to insert the device serial is using the TCP data source type. The serial I can insert there of the logger is, very different from the inverter serial and upon insertion of the serial, it gets corrected to a 9 digit version. It is clearly the wifi/logger serial I'm inserting. I used the serial I found on the solarman webpage. I could check if this is really the same serial of the wifi/logger.

Videodok, 2022-06-17:
Maybe it's just impossible for my model: 

"Some older Omnik inverters doe not have a possibility to get the data directly, some types do, but if not the portal seems the only place you can collect the submitted data. My inverter (installed October 2012) is of that older type, and cannot be accessed directly to get the actual data. "
https://jbsoft.nl/site/omnik-datalogger/

robbinjanssen, 2022-06-20:
Seems like that could be the case 😢 I'm afraid we can't help you, maybe the omnik-datalogger plugin works for you?

closed 2022-07-11 by github-actions[bot]