robbinjanssen/home-assistant-omnik-inverter #119: Unknown omnik fields in Homeassistant

mark0100, 2022-06-22:
The fields "inverter temperature" and "Solar production uptime" show a value of Unknown in Homeassistant after i added this integration.
Other fields are shown correctly:

[image]

Here is the javascript output from my inverter:

`var version="H4.01.36Y1.0.05W1.0.04";var m2mMid="605677026";var wlanMac="AC:CF:23:12:90:98";var m2mRssi="100%";var wanIp="192.168.2.1";var nmac="ACCF2312909B";var fephy="off";var webData="NLDN3020138T1242,NL1-V1.0-0061-4,V2.0-0024,omnik3000tl ,3000,1837,1264,261313,,4,";function initPageText(){var A=window.parent.reList("status");fileText("st1",A["t1"]);fileText("st2",A["t2"]);fileText("st3",A["t3"]);for(var B=1;B<=27;B++){if(B!=14){fileText("tx"+B,A[B])}}init_main_page();initStatus();changeFont();child_getH()}function upfold(A){if(document.getElementById("up_"+A+"_div").style.display=="none"){show("up_"+A+"_div");document.getElementById("up_"+A).src="image/up.png"}else{hide("up_"+A+"_div");document.getElementById("up_"+A).src="image/down.png"}if(A==3){if(document.getElementById("ser_status").src==""){document.getElementById("ser_status").src="server_status.html"}}}function init_main_page(){var E=document.form_cover_setting;cfg2Form(E);var C;var D;var B;var A;B=window.parent.reTip("1");A=window.parent.reTip("2");document.getElementById("cover_ver").innerHTML=version;document.getElementById("cover_mid").innerHTML=m2mMid;if(E.wifi_on.value=="disable"){document.getElementById("cover_ap_status").innerHTML=A;document.getElementById("cover_ap_mac").innerHTML="";document.getElementById("cover_ap_ssid").innerHTML="";document.getElementById("cover_ap_ip").innerHTML="";document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}else{if(E.sta_enable.value=="1"){document.getElementById("cover_ap_status").innerHTML=B;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=B;document.getElementById("cover_sta_rssi").innerHTML=m2mRssi;document.getElementById("cover_sta_ip").innerHTML=wanIp;C=wlanMac.substring(0,16);D="0x"+wlanMac.charAt(16);D=D*1+1;D=D.toString(16);D=D.toUpperCase();C=C+D;document.getElementById("cover_sta_mac").innerHTML=C;document.getElementById("cover_sta_ssid").innerHTML=E.sta_ssid.value}else{document.getElementById("cover_ap_status").innerHTML=B;document.getElementById("cover_ap_mac").innerHTML=wlanMac;document.getElementById("cover_ap_ssid").innerHTML=E.ap_ssid.value;document.getElementById("cover_ap_ip").innerHTML=E.lan_ip.value;document.getElementById("cover_sta_status").innerHTML=A;document.getElementById("cover_sta_rssi").innerHTML="";document.getElementById("cover_sta_ip").innerHTML="";document.getElementById("cover_sta_mac").innerHTML="";document.getElementById("cover_sta_ssid").innerHTML=""}}}function init_webdata_form(){var E=webData;var B="";var A,C,D;A=E.length;C=0;D=0;while(C<A){if(E.charAt(C)==","){C+=1;D+=1;web_data_show(D,B);B=""}else{B=B+E.charAt(C);C+=1}}}function web_data_show(D,A){var C=new Array("webdata_sn","webdata_msvn","webdata_ssvn","webdata_pv_type","webdata_rate_p","webdata_now_p","webdata_today_e","webdata_total_e","webdata_alarm","webdata_utime");if(D>10||D<=0){return}var B=document.getElementById(C[D-1]);if(A==""&&B!=null){B.innerHTML="---";return}if(B!=null){if(D==5){B.innerHTML=A+" W"}else{if(D==6){B.innerHTML=A+" W"}else{if(D==7){sv=A*1/100;B.innerHTML=sv.toString()+" kWh"}else{if(D==8){sv=A*1/10;B.innerHTML=sv.toString()+" kWh"}else{if(D==10){if(document.getElementById("webdata_sn").innerHTML=="---"){B.innerHTML="---"}else{B.innerHTML=A+window.parent.reTip("5")}}else{B.innerHTML=A}}}}}}else{alert(C[D])}};`

I added this integration manually to homeassistant as described in the README.

I am using Home Assistant Core 2022.6.6 and Frontend-versie: 20220601.0 - latest

MarijnS95, 2022-06-22:
This information is only available on the TCP backend, hence it's "onbekend" :wink: 

Perhaps we should document this more explicitly or disable these entities by default if not using the TCP backend - though it's a good way to remind users to try the TCP backend instead.

MarijnS95, 2022-06-22:
Not solved per-se, I should have done a better job of documenting this :)

mark0100, 2022-06-22:
You are forgiven my friend :-)