robbinjanssen/home-assistant-omnik-inverter #120: 1. Request reads 0 total_energy, leading to overshoot when actual values are read

tilchl, 2022-06-23:
## Detailed description
The HA integration has a problem with the inverter "waking up". I belive the first request reads 0 kWh for Total and Today production. [image]

This in turn leads for the Energy Dash to display the second request as a plus of total. 
[image]

Normally I could use the `solar_energy_today` but I disabled all communications between the logger and china.
That way the logger does not get a new date to reset the `solar_energy_today`.

ideally a `solar_energy_total` of zero would be ignored if larger values are present. 
I don't know what db HA is using but it might support dropping queries with decreasing values for said key.
Otherwise a query of the last value is neccesary and only contitionally write the new value.

robbinjanssen, 2022-06-23:
I think this is related to #51, I had similar problems with my inverter but they magically resolved themselves 🤷 
By caching the value and resetting it at 00:00, it should be good I think?

klaasnicolaas, 2022-06-23:
Strange, we once made an exception that pops up when your day and total indicate 0 kwh. Precisely to prevent us from getting this kind of situation, I'm curious what the logs are when this happens 🤔

tilchl, 2022-06-23:
I have already tried to analyze the TCP requests comming from the logger. But no luck.

I just dont want the logger to call home. I find it kind of alarming that it sends all meta information to the Solarmann API just by connecting it to a WiFi network.

Shame they dont share their modbus/TCP structure.

closed 2022-07-16 by github-actions[bot]