robbinjanssen/home-assistant-omnik-inverter #122: Overflow

FrankyB1983, 2022-07-17:
Sometimes I get strange overflows. For example. It work totally fine for a couple of days. And then I get a current power value of 76000 watts. My solar panels can only produce 1080 watts.

FrankyB1983, 2022-07-17:
[image]

robbinjanssen, 2022-08-08:
Go to you integrations, find the Omnik integration and click on the 3 dots. Then click on the "Download diagnostics" link, this file will provide all information :-)

SebastianMornhinweg, 2022-10-28:
Happens sometimes for me too. The daily count will be taken as current production. So in HA i then see for this hour a high peak and the daily production is doubled. I use HTML with a Deye/Bosswerk MI 600.
[image]

robbinjanssen, 2023-02-13:
We haven't received any input on this issue. Feel free te reopen but please provide the requested diagnostics.

closed 2023-02-13 by robbinjanssen