robbinjanssen/home-assistant-omnik-inverter #123: Very often I get unknown values

Dominion0815, 2022-08-03:
Hi,

in my screenshots you see that my Bosswerk MI600 often has gaps with unknown values.
Is it possible to cache the last state insteat?

[image]
[image]

Greetings
dominion

MarijnS95, 2022-08-03:
Could be similar to (duplicate of) https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/100?

Dominion0815, 2022-08-03:
oh yeah it is similar sorry.

My inverter is always accessible via wifi, only sometimes it does not output any values but displays "---" in the web interface.
I don't know if there is a query limit and a cache might help.

closed 2022-08-25 by github-actions[bot]