robbinjanssen/home-assistant-omnik-inverter #126: read alert field

CFenner, 2022-09-05:
I would like to see the `alert` field in home assistant. Any doubts about adding this?

## Detailed description

I would like to read `webdata_alarm` and make it available as a field in HA.

```
var webdata_total_e = "0.3";
var webdata_alarm = "F13";
var webdata_utime = "1";
```

## Context

This would be nice to have for diagnostics.

robbinjanssen, 2022-09-06:
I see the same field in my JS source, so we can probably add it. We need to add it in the [`python-omnikinverter`](https://github.com/klaasnicolaas/python-omnikinverter/blob/main/omnikinverter/models.py#L172-L181) library first though :)

klaasnicolaas, 2022-09-06:
We are working on it, it's just a puzzle to find out per source_type where this value is.

MarijnS95, 2022-09-06:
@CFenner Do you have a mapping of these error codes? According to certain Omnik docs like https://download.solarclarity.nl/2015/04/UserManual-OMNIK-3K4K5K-TL2.pdf F13 means that the device is overheating...

CFenner, 2022-09-06:
No, I don't have a mapping. It's definitely not overheating, current production is very low.

MarijnS95, 2022-09-06:
@CFenner How useful is this field without knowing what's up?

I can see use in knowing the value, but wouldn't want to unnecessarily scare users because of some non-empty error code/value.

PR #149 (2023-02-13): Add alarm code sensor to inverter
### What does it do?
Add a sensor with the alarm code status of the inverter

Closes #126
closed 2023-02-13 by robbinjanssen