robbinjanssen/home-assistant-omnik-inverter #127: add Connectivity Status Binary Sensor

CFenner, 2022-09-06:
I would like to see if my inverter is up & connected. 

## Detailed description

There is a `device_class: connectivity` which can be used to display this. 

[image]
[image]
[image]

Meanwhile is solved this by adding a template sensor:

```yaml
name: Bosswerk Veranda Status
state: >-
  {% if states('sensor.bosswerk_veranda_solar_current_power') in ("unavailable", "unknown") -%}
    off
  {%- else -%}
    on
  {%- endif %}
device_class: connectivity
```

## Context

Use in automations.

robbinjanssen, 2022-09-07:
So we could do this by checking "do receive data? e.g. status 200 OK?" -> UP, if not -> DOWN.
Is that something you could work with?

MarijnS95, 2022-09-07:
Is this `unavailable`/`unknown` value thought up by HASS when the integration hasn't provided any value for said entity for some time? We were thinking about driving this connectivity status by a configurable timeout on the requests that are sent by this integration (i.e. we won't get any http 200 replies / data if the inverter is offline/disconnected - if that happens for a couple requests in a row the inverter is likely unreachable).

CFenner, 2022-09-12:
> Is this `unavailable`/`unknown` value thought up by HASS when the integration hasn't provided any value for said entity for some time? 

I think that is how it works. The idea I had is that with a `connectivity` entity the status is even displayed when the device is not reachable.

CFenner, 2022-12-08:
now during winter time my inverter is always offline. I would expect the integration to display this as the entities being unavailable. Also having a status entity would top here. 

[image]

robbinjanssen, 2023-02-04:
@CFenner does it show the image above after a restart of HA or a reload of the integration?

CFenner, 2023-02-04:
Both

closed 2023-02-13 by robbinjanssen