robbinjanssen/home-assistant-omnik-inverter #132: one of my 2 inverters stopped transmitting data and is even not accessable anymore

htelgen, 2022-09-27:
I have 2 inverters: Omnik 3000tls and Omnik 1000tls. Because i have 15 pannels the 3000 wasn't enough. A couple of weeks ago the 3000tls was not available on HA anymore. It still has the local IP-address but is not responding to webcall. It might be a coincidant but at that time I also connected the inverters to the Solarmanagement app.

## Detailed description
I tried to make ik set-up an accesspoint again with no result

## Context
I was used to read the solar data directly instead of from the P1 meter. The P1 isn't giving the clean production data, only what is left after own use.

## Diagnostics (the 3000 I deleted to try it again connecting)
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2022.9.7",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.5",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.61-v8",
    "supervisor": "2022.09.1",
    "host_os": "Home Assistant OS 9.0",
    "docker_version": "20.10.17",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "omnik_inverter": {
      "version": "2.3.0",
      "requirements": [
        "omnikinverter==0.7.0"
      ]
    },
    "hacs": {
      "version": "1.24.5",
      "requirements": [
        "aiogithubapi>=21.11.0"
      ]
    }
  },
  "integration_manifest": {
    "domain": "omnik_inverter",
    "name": "Omnik Inverter",
    "config_flow": true,
    "version": "2.3.0",
    "documentation": "https://github.com/robbinjanssen/home-assistant-omnik-inverter",
    "issue_tracker": "https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues",
    "codeowners": [
      "@robbinjanssen",
      "@klaasnicolaas"
    ],
    "requirements": [
      "omnikinverter==0.7.0"
    ],
    "iot_class": "local_polling",
    "is_built_in": false
  },
  "data": {
    "entry": {
      "title": "Huis",
      "data": {
        "host": "**REDACTED**",
        "source_type": "javascript"
      },
      "options": {
        "scan_interval": 4
      }
    },
    "data": {
      "device": {
        "signal_quality": 29,
        "firmware": "H4.01.38Y1.0.08W1.0.07",
        "ip_address": "192.168.2.59"
      },
      "inverter": {
        "serial_number": "NLBN1020154P8039",
        "model": "omnik1000tl2",
        "firmware": null,
        "firmware_slave": "V2.2-91710",
        "solar_rated_power": 1000,
        "solar_current_power": 214,
        "solar_energy_today": 0.4,
        "solar_energy_total": 6322.2
      }
    }
  }
}

```json
"Copy here the code or upload the file"
```

klaasnicolaas, 2022-09-27:
It is not clear to me what really the problem is and to what extent this is still an integration problem for Home Assistant or another third party issue 🤷🏻‍♂️

htelgen, 2022-09-27:
I cut it of power and now having initial connection again

htelgen, 2022-09-27:
I got it back on showing in HA again. So, I'm good.

closed 2022-09-27 by klaasnicolaas