robbinjanssen/home-assistant-omnik-inverter #137: Add Host URL sanitizing

dirk2947, 2022-10-30:
## Detailed description

When adding a URL by copying it from a webbrowser it also adds the http;//xxx.xxx.xxx.xxx in front (Firefox in my case). It would be nice if the Integration could autodetect this and remove the protocol or throw a more telling error explaining that the IP should be input as numbers only.

## Context

It didn't take me long to realize where I messed up, but I think it might reduce some minor confusion.

klaasnicolaas, 2022-10-30:
My preference would be to throw an error, this to prevent us from teaching user behavior that is already wrong by default.
In the end, we are clearly asking for a HOST and not a URL.

dirk2947, 2022-10-30:
> My preference would be to throw an error, this to prevent us from teaching user behavior that is already wrong by default. In the end, we are clearly asking for a HOST and not a URL.

Yup that would be great and it would have the added benefit of cutting down some minor confusion in the future between host / url for people that aren't too tech savvy.

klaasnicolaas, 2022-10-30:
Have something in the works to fix this:

[image]

closed 2022-10-31 by robbinjanssen