robbinjanssen/home-assistant-omnik-inverter #139: Unavailable status

AlbertHakvoort, 2022-11-15:
Is it possible to prevent the unavailable status in Home Assistant, so when the connection is lost to the inverter, the last known values are showed

[image]

MarijnS95, 2022-11-15:
This is however the right value to propagate when it comes to how HASS stores these values in the `states` table and in the short/long-term statistics table.

I recommend using cards that base their values on statistics rather than the raw entity states (even if these are only updated every hour) or look for cards / template values that display the last known state for an entity instead of the most recent unknown value.

robbinjanssen, 2022-11-30:
Hi @AlbertHakvoort, this could be fixed by adding an (optional) caching layer. We have an issue for this already, this is a duplicate of https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/51

closed 2022-11-30 by robbinjanssen