robbinjanssen/home-assistant-omnik-inverter #142: omnik inverter tcp fails to configure in Home-assistant

janknol, 2022-12-30:
intergration of the  omnik inverter into Home Assistant 

## Detailed description
omnik inverter 3000TL (2013)
add intergration
omnik
pop up 
choose TCP
name solar
ip address
serial 0

after a while an error pops up:

Buffer size too small (85 instead of at least 125 bytes)

## Context

Why is this change important to you? 
I want to see the solar production in Home Assistant dashboard

How can it benefit other users?

-->

MarijnS95, 2022-12-30:
Can you share a dump of the returned data, typically captured with wireshark? This might be caused by the inverter sending the data split in a weird way, something I have fixed locally but have not yet made a PR for.

janknol, 2022-12-30:
marijnS95:
i installed wire shark at my macbook, run de HA app at de macbook , start the integration, choose omnik inverter, choose TCP as setup, ip address and serial , submit
started wireshark capture wifi en0 and looking in the capture for the inverter's ip address , not found
i have another pi where  the omnik data capture software  from github  is installed  , it  poll's the inverter's 8899 port and saves this data in a mysql database.
which proofs that the data could be collected from the inverter
what is the HA implementation doing different?

MarijnS95, 2022-12-30:
> marijnS95: i installed wire shark at my macbook, run de HA app at de macbook , start the integration, choose omnik inverter, choose TCP as setup, ip address and serial , submit started wireshark capture wifi en0 and looking in the capture for the inverter's ip address , not found

Can you run it on the machine/interface that the HA instance is running on? What is a HA app "at de" macbook, isnt't that just the web interface?

> i have another pi where the omnik data capture software from github is installed 

There are many such projects so that doesn't mean much. You're probably referring to the most popular one which is https://github.com/jbouwh/omnikdatalogger?

> it poll's the inverter's 8899 port and saves this data in a mysql database. which proofs that the data could be collected from the inverter what is the HA implementation doing different?

It doesn't perform any data validation on the returned TCP packets, which is what I implemented at the core. However, it also doesn't process multiple "received streams" when the result ends up being segmented, so that is unlikely to be the cause. Perhaps this inverter is simply returning less fields, which are optional in the other logger (please confirm which one you are using) but mandatory in the [current parser](https://github.com/klaasnicolaas/python-omnikinverter/blob/5a5dcf74125a0372051ee113edf3c5eebf18ce96/omnikinverter/tcp.py#L49-L71).

---

If you can't get wireshark to work we can dump the raw packet from within `python-omnikinverter` code, too.

janknol, 2022-12-30:
I use the omnik-data-logger --> github.com/Woutrrr/Omnik-Data-Logger.git

>we can dump the raw packet from within python-omnikinverter code< 
how do i do that?

MarijnS95, 2022-12-30:
> I use the omnik-data-logger --> github.com/Woutrrr/Omnik-Data-Logger.git

Right, a different one after all. This [also doesn't read packets more than once](https://github.com/Woutrrr/Omnik-Data-Logger/blob/526085c25cfe2d9ed216f0094b6e01ab54c5a175/OmnikExport.py#L73) but it could be more lenient in how big of a message it expects.
 
> > we can dump the raw packet from within python-omnikinverter code
>
> how do i do that?

https://github.com/klaasnicolaas/python-omnikinverter/blob/5a5dcf74125a0372051ee113edf3c5eebf18ce96/omnikinverter/omnikinverter.py#L155 dump the value of `raw_msg` there to a file, with something like:

```python
with open("3000TL", "wb") as f:
    f.write(raw_msg)
```

janknol, 2023-01-17:
hi,

the issue can be closed, i managed to get the data from the inverter by using the hacs implementation from hein oldenhuis

MarijnS95, 2023-01-17:
This issue should not be closed. Hein's repo is unmaintained and this is supposed to be its improved successor, and we intend to make usable for all inverters.

That said I did not received any of the requested info from you to aid debugging. Since it works with the other plugins I can only assume the data layout is the same but lacking the trailing padding and maybe some fields (beyond 85 bytes) that we can make optional. If you are not able to provide a data dump, would you at least be available to test a change that enables this? I'll describe how to set it up.

janknol, 2023-01-17:
ok, i wasn’t able to get a data dump.but i’ll test another setup when you have itregards,janSent from my iPhone

MarijnS95, 2023-01-18:
@janknol please don't use email replies or at the very least carefully watch the newlining, GitHub makes these things atrocious to read 🫣.

---

Looking at the datastructure we go down from 125 to 85 bytes by exactly stripping the last four fields - `firmware`, `firmware_slave` and their padding:

https://github.com/klaasnicolaas/python-omnikinverter/blob/9ce9ef715630068d295d85093ba1739bf874f981/omnikinverter/tcp.py#L52-L70

Either this is new and wasn't yet provided in your software version, or `padding1`/`unknown0`/`padding2` contain this information, but I cannot know that without receiving any sort of data dump from you @janknol.

I can see about providing a patch later that ignores these fields and also does the dumping for you.

PeeVv, 2023-04-13:
Same issue here with my Omnik 3000TL. The other Omnik integration works, but I get the same "Buffer size too small (85 instead of at least 125 bytes)" error. Attached is a wireshark dump from the packages between my Home Assistant server and my inverter during the setup. Please let me know if this is the correct data, I'm not really knowledgeable about wireshark.

[Omnik wireshark filtered.pcapng.zip](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/11225324/Omnik.wireshark.filtered.pcapng.zip)

closed 2023-05-29 by klaasnicolaas

PeeVv, 2023-06-15:
Is this already implemented? I still get the same error (Buffer size too small (85 instead of at least 125 bytes) with the latest version 2.6.1.

MarijnS95, 2023-06-15:
@PeeVv we might have to cut a new release of `python-omnikinverter` and `home-assistant-omnik-inverter` for that to trickle through.