robbinjanssen/home-assistant-omnik-inverter #143: Inverter temperature reading becomes 6,532.6 when 3000TL is not providing power

imredamen, 2023-01-04:
I am using version 2.4.2 (although the manifest still says 2.4.0) of the integration. I am using the TCP connection to my Omniksol 3000TL inverter. I was having a look at the registered inverter temperature and it defaults to 6,532.6 during the night. 

## Detailed description
Bug/Enhancement - have inverter temperature sensor default to 0 when no input is received from the inverter. Based on the pattern I assume the value is not present when the inverter is not active. Somehow currently this defaults into a wrong value of 6,532.6, where 0  would be preferred. 

Another small thing is that the version in the manifest file (2.4.0) does not match the release version (2.4.2) - having the correct version would allow to check for newer versions easily. 

## Context
tracking inverter temperature during the day, without the high value all other variations and patterns become visible.

## Diagnostics

[config_entry-omnik_inverter-750a11b75890ff7d00b44604b24feb83.json.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/10344561/config_entry-omnik_inverter-750a11b75890ff7d00b44604b24feb83.json.txt)

robbinjanssen, 2023-01-05:
Hi, I'm currently working on another integration which is taking up most of my free time, but once im done i'll give some love to this integration.

Do you know if your inverter is online at that moment? Or does the connection actually fail?

imredamen, 2023-01-06:
As far as I know the inverter is online from a network perspective (tcp, http calls). 
Having another look at the data, I notice the values of the other sensors remaning unchanged or default to 0 in the same timeframe. So, the inverter seems down from a functional perspective. 

There is an error occuring in the logs, but not specific to the sun down times:

2023-01-06 09:19:12.666 ERROR (MainThread) [custom_components.omnik_inverter] Unexpected error fetching omnik_inverter data: coroutine ignored GeneratorExit
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 225, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/omnik_inverter/__init__.py", line 125, in _async_update_data
SERVICE_INVERTER: await self.omnikinverter.inverter(),
File "/usr/local/lib/python3.10/site-packages/omnikinverter/omnikinverter.py", line 186, in inverter
fields = await self.tcp_request()
RuntimeError: coroutine ignored GeneratorExit

MarijnS95, 2023-01-11:
@imredamen can you possibly perform a wireshark capture of this network traffic (a single request and reply is enough) and share it with us?

Or at the very least share a diagnostic while this odd temperature value is showing; the current diagnostic contains `"temperature": 24.5,` and looks to have been taken under normal operation (inverter delivering 107W).

> have inverter temperature sensor default to 0 when no input is received from the inverter.
> [...]
> As far as I know the inverter is online from a network perspective (tcp, http calls).

It is very important to know whether this value is a reply from the inverter, which it very likely is as we hardly do any processing or caching on these values (it would either be unknown/unavailable, or the last known value, not suddenly jump to a random value entirely). The value is a 16-bit unsigned integer, that we multiply by 0.1 to get a proper temperature reading, so the original value was `0xff2e` which "may mean something special" (i.e. "hardware turned off, no sensible temperature reading available").

---

The `GeneratorExit` is afaik a badly represented timeout, but [I've given up on fixing it properly](https://github.com/klaasnicolaas/python-omnikinverter/pull/207).

imredamen, 2023-01-11:
hi @MarijnS95 ,
Attached a diagnostic file with the value after inverter stopped producing energy. 
I've also attached 2 tcpdump files, one when I get an expected value (active), the other (sundown) with the unexpected high value. Hope this helps, let me know if you need anything else!

[sun-down-config_entry-omnik_inverter-750a11b75890ff7d00b44604b24feb83.json.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/10394160/sun-down-config_entry-omnik_inverter-750a11b75890ff7d00b44604b24feb83.json.txt)
[tcpdump-omnik-active.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/10394161/tcpdump-omnik-active.txt)

[tcpdump-omnik-sundown.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/10394159/tcpdump-omnik-sundown.txt)

MarijnS95, 2023-01-11:
@imredamen I'm not seeing that value anywhere in your sundown dump. Take for example this snippet:

```
	0x0040:  8102 014e 4c42 4e33 3032 3031 3534 5035  ...NLBN3020154P5
	0x0050:  3035 3101 050d ce0c 9200 0000 0200 0100  051.............
```

If we follow [the datalayout](https://github.com/klaasnicolaas/python-omnikinverter/blob/5a73f4f39956a611a471d9bbf64076d8fb298e22/omnikinverter/tcp.py#L53-L54) the temperature is a big-indian unsigned 16-bit integer right after the serial number. The serial number is `NLBN3020154P5051`, and the bits right after that are `01 05`. This corresponds to `0x105 = 261 = 26.1⁰C`. All other occurrences are in the same ballpark.

Are you sure the plugin was displaying garbage values at this point? We might want to insert some extra logging in the plugin otherwise... Nowhere have I found the aforementioned `0xff2e`...

imredamen, 2023-01-12:
@MarijnS95 - I am sure that I captured data with and without the values, when I look at my files the look the same. So, I probably messed up something there...

So another attempt:
[omnik-imre-sundown-20230112.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/10399263/omnik-imre-sundown-20230112.txt)

        0x0050:  014e 4c42 4e33 3032 3031 3534 5035 3035  .NLBN3020154P505
        0x0060:  31ff 2e00 0000 0000 0000 0000 0000 0000  1...............

Here you can see ff 2e is actually sent by the inverter. So, the plugin seems to show the value the inverter actually sends. Still this value makes no sense in monitoring the inverter...

robbinjanssen, 2023-02-13:
This is fixed in the next release.

closed 2023-02-13 by robbinjanssen