robbinjanssen/home-assistant-omnik-inverter #152: Error occurred while communicating with Omnik Inverter device

Buffer379x, 2023-02-15:
## Detailed description
Hallo Robbin, Hallo everybody,

After updating to the most recent version this morning the inverter integration is not working anymore for me. Error message that is being shown on the integration page is:

"Retrying setup: Error occurred while communicating with Omnik Inverter device"

Since I did not find a way to access the configuration page I uninstalled and reinstalled the integration, but the error message stays the same. A manual connection to the inverter is still possible (Bosswerk MI600).

Log file from Home Asisstant shows the following entry:

Logger: homeassistant.config_entries
Source: config_entries.py:1211
First occurred: 09:37:54 (1 occurrences)
Last logged: 09:37:54

Config entry 'Balkonkraftwerk' for omnik_inverter integration not ready yet: Error occurred while communicating with Omnik Inverter device; Retrying in background

## Context
-

## Diagnostics

2023-02-15 09:51:03.423 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Balkonkraftwerk' for omnik_inverter integration not ready yet: Error occurred while communicating with Omnik Inverter device; Retrying in background

(not sure if this is helpful or if I am missing some other diagnostics log file)

robbinjanssen, 2023-02-15:
You can downgrade to the previous version for now, can you share a diagnostics file? You can find it under the 3 dots at the integration file. What source do you use?

Buffer379x, 2023-02-15:
Thank you a lot for the hint, downgrading as the easiest solution for now did not come into my mind... :-)

v2.4.2 works out of the box again. Strange enough I am not able to download the diagnostics from v2.5.0 (see screenshot). I am still attaching the ones from v2.4.2, but I doubt they will be of help since that version is working.

Not sure what you mean with source:
Source for scraping the data is HTML (Inverter = Bosswerk MI-600)
Source for the integration = HACS

[config_entry-omnik_inverter-804f25f3eee954e2888dcff1472d70a3.json.txt](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/10743035/config_entry-omnik_inverter-804f25f3eee954e2888dcff1472d70a3.json.txt)

[image]

robbinjanssen, 2023-02-16:
@Buffer379x do you think it is possible to open the connection to your inverter for me? Forward the port on your router and send me the details :-) You can find my email on my github profile.

It's kinda hard to debug otherwise

Buffer379x, 2023-02-16:
@robbinjanssen You've got mail :-)

timn, 2023-02-16:
I'm experiencing the same problem (also Bosswerk MI-600). Happy to share an HTML dump of the inverter if that helps.

robbinjanssen, 2023-02-16:
@timn @Buffer379x gave me access to his inverter, so when it is sunny outside tomorrow i'll have a look :-)

closed 2023-02-17 by robbinjanssen

robbinjanssen, 2023-02-17:
@timn try upgrading to v2.5.2, this resolved the issue for @Buffer379x :)

timn, 2023-02-17:
I can confirm that 2.5.2 fixes the problem. Thank you for the quick fix, @robbinjanssen! Out of curiosity, what is the root cause of the problem, why doesn't the HA function work?

MarijnS95, 2023-02-19:
> @klaasnicolaas, @MarijnS95 and I are discussing whether or not we should combine the information in a single call. I _think_ that will solve the problem for the bosswerk inverters as well :)

Yes! I've submitted the necessary changes for that now (https://github.com/klaasnicolaas/python-omnikinverter/pull/340, https://github.com/robbinjanssen/home-assistant-omnik-inverter/pull/157), sorry for the wait!

PR #155 (2023-02-17): Do not pass session from home assistant, fails for some inverters
### What does it do?
When using the latest version with a bosswork inverter, the second call (getting the device information) might end up in a server disconnected exception. For now we'll let the `python-omnikinverter` package handle all the sessions.

Fixes #152