robbinjanssen/home-assistant-omnik-inverter #153: New devices and entities generated after upgrade.

ronaldt80, 2023-02-15:
Upgrade to v 2.5 results in new entity IDs being created wrecking the energy dashboard history.

Following the upgrade to the new version my old entities were listed unavailable with new entities being created. This breaks historic data in the energy dashboard. When i changed the names all history was gone. Now rolled back to yesterday's backup to restore data.

Could someone please advise how to perform the upgrade without this impact?

## Context

## Diagnostics

```json
"Copy here the code or upload the file"
```

robbinjanssen, 2023-02-16:
Did you come from version v2.4.x? or an earlier version?

robbinjanssen, 2023-02-16:
I think this has to do because i've used another unique_Id, i though they were similar:
v2.4.2:
https://github.com/robbinjanssen/home-assistant-omnik-inverter/blob/v2.4.2/custom_components/omnik_inverter/sensor.py#L221

v2.5.0
https://github.com/robbinjanssen/home-assistant-omnik-inverter/blob/main/custom_components/omnik_inverter/sensor.py#L242

The name however is the same

robbinjanssen, 2023-02-16:
@ronaldt80 Try v2.5.1 :)

closed 2023-02-16 by robbinjanssen

ronaldt80, 2023-02-17:
Perfect! This works - many thanks for the quick follow up

PR #154 (2023-02-16): Reuse unique names from previous version
### What does it do?
Reuse the naming for entity_ids like it was done in v2.4.2

Closes #153