robbinjanssen/home-assistant-omnik-inverter #158: Update interval in seconds

serrj-sv, 2023-02-20:
Hi! My inverter supports TCP mode and I'd like to query it more frequently. Could you make it possible to set minimum update interval in seconds instead of minutes? I believe this cam be done by replacing minutes with seconds here: https://github.com/robbinjanssen/home-assistant-omnik-inverter/blob/main/custom_components/omnik_inverter/coordinator.py#L52

robbinjanssen, 2023-02-21:
Yes we can do that, I have to find out how to migrate the "previous" config from minutes to seconds though.

MarijnS95, 2023-02-21:
FYI I'm working on a change that keeps the TCP connection open, where the inverter will send an update every minute (observed on a 3000TL) but have not yet found a way to change that rate.

Perhaps we should keep that as a separate configurable mode, to support both `local_polling` (also needed for the JS/HTML variants) and `local_push`?

GelsemaP, 2024-02-22:
isnt the code missing to store the time between updates in config? in strings i dont see the config key for timedelta.
also it seems that supporting seconds instead of minutes is just a matter of changing the timedelta to seconds. only concern is that with an update minutes are suddenly seconds, which you could fix by adding code for an update that rewrites the value. perhaps even add an extra field which allows you to use the timedelta as variable (minutes/seconds) as dropbox and use that for your timedelta command. I'm not that good in coding so cant try myself... sorry...

closed 2026-02-16 by robbinjanssen