robbinjanssen/home-assistant-omnik-inverter #193: Failed to connect to the Omnik

lukasmedia, 2023-05-01:
2023-05-01 09:56:07.329 ERROR (MainThread) [custom_components.omnik_inverter.coordinator] Failed to connect to the Omnik
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/omnikinverter/omnikinverter.py", line 90, in request
    response = await self.session.request(
  File "/usr/local/lib/python3.10/site-packages/aiohttp/client.py", line 536, in _request
    conn = await self._connector.connect(
  File "/usr/local/lib/python3.10/site-packages/aiohttp/connector.py", line 540, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/connector.py", line 901, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/connector.py", line 1175, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
  File "/usr/local/lib/python3.10/site-packages/aiohttp/connector.py", line 980, in _wrap_create_connection
    return await self._loop.create_connection(*args, **kwargs)  # type: ignore[return-value]  # noqa
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1060, in create_connection
    sock = await self._connect_sock(
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 969, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.10/asyncio/selector_events.py", line 501, in sock_connect
    return await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/omnikinverter/omnikinverter.py", line 89, in request
    async with async_timeout.timeout(self.request_timeout):
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.10/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/omnik_inverter/coordinator.py", line 92, in _async_update_data
    SERVICE_INVERTER: await self.omnikinverter.inverter(),
  File "/usr/local/lib/python3.10/site-packages/omnikinverter/omnikinverter.py", line 183, in inverter
    data = await self.request("js/status.js")
  File "/usr/local/lib/python3.10/site-packages/omnikinverter/omnikinverter.py", line 99, in request
    raise OmnikInverterConnectionError(
omnikinverter.exceptions.OmnikInverterConnectionError: Timeout occurred while connecting to Omnik Inverter device

robbinjanssen, 2023-05-01:
Hi @lukasmedia, it seems like your HA instance cannot reach your inverter. Can you try a ping or web-request from your HA instance?

lukasmedia, 2023-05-01:
Thx @robbinjanssen  

Most of the time HA can connect to the inverter without problems .. it seems:

[image]

And ping also works fine:

[image]

But the error messages are really often; every few minutes or so. It looks like the actual status is parsed; but as a bonus an error message is generated.

robbinjanssen, 2023-05-01:
Looking at your ping (1000ms = 1 sec) the problem could very well be an unstable connection to your inverter. Can you keep your ping open and see when the error occurs, if the ping also has a timeout?

lukasmedia, 2023-05-01:
@robbinjanssen i changed the IP from dhcp to static in the wireless config and configured the gateway & dns to my router IP. Now the ping around 2 ms. (the wifi router is 40 cm. next to the omnik inverter). So ping improved; lets wait for the error messages.
I have a suspect in my network: pihole.

robbinjanssen, 2023-05-03:
@lukasmedia any update? :)

closed 2023-05-15 by github-actions[bot]