robbinjanssen/home-assistant-omnik-inverter #196: Issue after updating to HA 2023.5

dheijt, 2023-05-05:
Hi,

The Omnik integration is no longer working after updating HA to version 2023.5.

Error message in log:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/omnik_inverter/__init__.py", line 47, in async_setup_entry
    hass.config_entries.async_setup_platforms(entry, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'

Seems that other custom integrations have the same issue: 

https://community.home-assistant.io/t/2023-5-lets-talk/567331/21
https://github.com/garbled1/homeassistant_ecowitt/issues/149

Thanks.
Dennis

Gilbertonl, 2023-05-06:
I have same issue. It stopped working after the update

Gilbertonl, 2023-05-06:
After updating omnik to newest version it's working for me again :)

klaasnicolaas, 2023-05-06:
You are probably some releases behind on the integration, because this has already been solved in version 2.5.1 and the latest version is already 2.6.0. So I advise you to update the integration 😉

closed 2023-05-06 by klaasnicolaas

dheijt, 2023-05-07:
Updated the integration and after that HA and all is working fine. 

Thanks