robbinjanssen/home-assistant-omnik-inverter #199: I don't get all stats from the Inverter

molekuul, 2023-05-12:
Hi I'm new to your code, so maybe I missed something..
I installed your code manually in homeassistant, and configured the IP from my Omnik 2000TL as a JS  system.
I do recieve several data fields, however there were hidden Entities, I unhide them, but I Don't see the value of these entities.
like Temperature is Unknown

Previously I used "Omnik-Data-Logger" from https://github.com/Woutrrr/Omnik-Data-Logger
where I do see the values:

python ./LiveStats.py
2023-05-12 11:18:22,448 DEBUG Importing output plugin ConsoleOutput
2023-05-12 11:18:22,448 INFO connecting to **.**.**.** port 8899
2023-05-12 11:18:23,109 INFO ID: NLDN2020******
2023-05-12 11:18:23,109 DEBUG Run pluginConsoleOutput
ID: NLDN2020*******
E Today:  1.25   Total: 16370.5
H Total: 33307   Temp:  30.8 
PV1   V: 237.4   I:  1.4
PV2   V:   0.0   I: 17.9
PV3   V:    -1   I:   -1
L1    P:   314   V: 227.5   I:  1.3   F: 50.05
L2    P:    -1   V:    -1   I:   -1   F:    -1
L3    P:    -1   V:    -1   I:   -1   F:    -1

Did I miss something in my config, or are these entities nod included yet in your code, in the config for the Omnik-Data-Logger, I needed my serial number to be able to retrieve the data.

kind reragds.
Joris Weijters

MarijnS95, 2023-05-12:
Values like temperature, per-string voltage/amperage and per-output statistics are only available via tue TCP (not JS) data source, which is what Omnik-Data-Logger also uses under the hood.

molekuul, 2023-05-12:
Thanks, I now configured this, and I see these values.
however, The device values for the IP number and the signal Quality states unknown, 
and when I try to adjust time between entity updates I get an error when trying to save the configuration:

**User input malformed: expected str for dictionary value @ data['serial']**

molekuul, 2023-05-12:
When In the config panel I put my serialnumber between ""  I don't get the error above. Maybe the config "thinks" my serialnumber is a Integer in stead of a string.

molekuul, 2023-05-12:
However when I put the serial number in between "" I get an error:
 ERROR (MainThread) [custom_components.omnik_inverter.coordinator] Unexpected error fetching omnik_inverter data: 'str' object has no attribute '
to_bytes'

MarijnS95, 2023-05-13:
> however, The device values for the IP number and the signal Quality states unknown,

These are _not_ provided via the TCP backend, it's either one or the other. I don't think Woutrrr's plugin provides these either?

> User input malformed: expected str for dictionary value @ data['serial']

That should really be a number type, but it looks like @raymanP set it to a `str` in the "reconfig" PR recently if I understand this correctly:

https://github.com/robbinjanssen/home-assistant-omnik-inverter/pull/188/files#diff-be9c8708f0b6f7f48716d7e82f7ac98072e1a1ba94d408bc38e63d7d21cb0e10R346-R350

raymanP, 2023-05-13:
Ooh yes, I didn't check the serial type and could test only the html variant.
Thanks for fixing @MarijnS95

closed 2023-06-13 by robbinjanssen