robbinjanssen/home-assistant-omnik-inverter #204: Bug: Connection timeouts in HA 2023.6.1

mxx-lxg, 2023-06-10:
## Detailed description

This is a bug that occures since i updated HA to  2023.6.1. I've updated yesterday and ever since i get random connection timeouts. I have 2 Bosswerk MI-300 inverters. They're definitely online and the admin page is reachable, but the integrations can't connect to them. It happens randomly throughout the day and it can take hours before the integrations can connect again.

## Diagnostics
[debug log file](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/11712202/home-assistant_omnik_inverter_2023-06-10T10-42-59.193Z.log)

MarijnS95, 2023-06-10:
Are you sure the inverters are properly, constantly reachable from your HA instance? According to the diagnostics this seems like a more low-level connection error that's not explicable by the HA upgrade alone.

Just in case, make sure you're on the latest version of this plugin. Or perhaps test out one of the other backends (though I don't know which of those are compatible with the BW).

robbinjanssen, 2023-06-12:
You could add a binary_sensor that pings your inverter every X seconds for now, and see if the connection timeouts also occur on the ping:

```yaml
binary_sensor:
  - platform: ping
    host: x.x.x.x
    name: "My Inverter"
    count: 2
    scan_interval: 5
```

mxx-lxg, 2023-06-12:
> Are you sure the inverters are properly, constantly reachable from your HA instance? According to the diagnostics this seems like a more low-level connection error that's not explicable by the HA [...]

There are indeed short WiFi outtages throughout the day, but those show up as small minute-long gaps in the data. The problem i'm having is that the integrations stop getting data for hours. Now that i think about it. It always happens some time after sunrise until shortly before sunset. Maybeeeee overheating might be an issue somewhere. I will try and move my garden access-point to a less sunny location and see what that does.

> You could add a binary_sensor that pings your inverter every X seconds for now, and see if the connection timeouts also occur on the ping:

I'm gonna try that too.

robbinjanssen, 2023-06-13:
Thanks for the update @mxx-lxg, i'll close this issue for now. If the problem still occurs after fixing the network "issues" please reopen :)

closed 2023-06-13 by robbinjanssen