robbinjanssen/home-assistant-omnik-inverter #211: Error after HA Update

alenet, 2023-06-25:
I own a Bosswerk mi600 and was using the Omnik over HTML to get the data to HA. Unfortunately after the latest update I am not able to get data from the micro inverter. 
I have access to the inverter and of course also can ping it. 

[image]

Basically I can add the inverter into HA but it got no entities. Do you have any idea what can be the problem?
Thanks for your help

robbinjanssen, 2023-06-26:
Is there any error in the HA log? Which version of HA do you use and which version of the integration?

alenet, 2023-06-26:
Hi Robbin,
This is the HA Version
[image]
This is the log file:
[home-assistant_omnik_inverter_2023-06-26T09-31-11.803Z.log](https://github.com/robbinjanssen/home-assistant-omnik-inverter/files/11866707/home-assistant_omnik_inverter_2023-06-26T09-31-11.803Z.log)

This should be the failure: 
2023-06-24 00:09:13.727 WARNING (MainThread) [homeassistant.config_entries] Config entry 'PVAnlage' for omnik_inverter integration not ready yet: Error occurred while communicating with Omnik Inverter device; Retrying in background

I dont know how to check the version of the integration. Can you help me there?

Thank you for your support  :)  As I´ve said, it worked so far as described in the installation but since the latest update this is happening. 
Best Regards,
Alejandro

robbinjanssen, 2023-06-26:
Hi, seems like you are using an old version of the integration which has functionality home assistant no longer supports.

Try updating to the latest version using HACS or copy/pasting manually :)

alenet, 2023-06-26:
Hi Robbin
[image]

I´ve installed HACS again using the terminal.

[image]

Restarted the Raspberry and the problem persist :(
Im not sure if I am doing everything right.

robbinjanssen, 2023-06-26:
You've updated HACS, not the custom integration. 

You need to open HACS in your home assistant, click on "integrations", click on the big blue button bottom right (Explore & Download repositories) and then search for the `Omnik Inverter Solar Sensor (No Cloud)` integration.

Or alternatively; update [manually](https://github.com/robbinjanssen/home-assistant-omnik-inverter#manual-installation)

alenet, 2023-06-26:
Problem Solved! Thank you Robbin!
Sorry for my incompetence :) 

PS: I will think twice next time I update HA. Everytime I do it something stops working.

robbinjanssen, 2023-06-26:
Just keep everything up to date and you'll be fine :) HA has gotten really mature of the last few years, read release notes and look for potential breaking changes. In my experience keeping everything up to date gives you less problems in the future! :-)

closed 2023-06-26 by robbinjanssen