robbinjanssen/home-assistant-omnik-inverter #220: Solar Energy Sensor (Day) provides weird values

NCO3, 2023-08-01:
## Detailed description

## Context

## Diagnostics

If any more information is needed, just let me know.
Thank you for your contribution and support.

robbinjanssen, 2023-08-01:
I see you are using the daily total as input for the dashboard sensor, may I recommend using the "total" production sensor? This is what I use and it works like a charm :)

NCO3, 2023-08-01:
Of course you may @robbinjanssen 😊
Much appreciated
I will give it a try and report back.

klaasnicolaas, 2023-08-01:
We do have to see what exactly happens to the data when the day value goes to 0. We can exclude this kind of behavior with an exception

NCO3, 2023-08-01:
Hi Klaas,
Sounds good.
Please let me know what exactly you need and I will prepare whatever I can.

robbinjanssen, 2023-08-03:
@NCO3 is your inverter unreachable (offline) or does it return `0` as a value?

NCO3, 2023-08-03:
Or seems that the value 0 is correlated to the reconnection of the inverter:
[image]
The time stamp of the connection is the very same like the value going to 0.
This is true for several other occasions of 0

klaasnicolaas, 2023-08-03:
@robbinjanssen we need to give him a custom version where every request data is logged, then we can see exactly what is happening.

closed 2023-09-10 by github-actions[bot]

NCO3, 2023-09-12:
All,
Of course it would be nice to be fixed officially, but I found a workaround.
Please see here:
https://community.home-assistant.io/t/help-interpreting-the-energy-dashboard/597581/16