robbinjanssen/home-assistant-omnik-inverter #223: Omnik2000tl AttributeError: 'ConfigEntries' object has no attribute 'async_forward_entry_setups'

Jampie85, 2023-08-11:
After adding i get this error ?
I dont know how to solve this error.

[image]
[image]

Logger: homeassistant.config_entries
Source: custom_components/omnik_inverter/__init__.py:29
Integration: Omnik Inverter ([documentation](https://github.com/robbinjanssen/home-assistant-omnik-inverter), [issues](https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues))
First occurred: 22:20:29 (1 occurrences)
Last logged: 22:20:29

Error setting up entry Casita for omnik_inverter
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/omnik_inverter/__init__.py", line 29, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_forward_entry_setups'

klaasnicolaas, 2023-08-11:
You are using an older version of the integration

Jampie85, 2023-08-12:
I was using the latest stable version, but seems that latest stable version is 2 years old lol.
Now i switched to 2023.4.6 and everything works perfect.
Thanks for the onmik integration :)

klaasnicolaas, 2023-08-12:
Latest stable version of the integration is 2.6.1 and 3 months old 🤷🏻

Jampie85, 2023-08-12:
Sorry, i was talking about the armv7 Home Assistant version.
2.6.1 not working with the latest stable version from docker Home Assistant.
Had to manual install 2023.4.6 Home Assistant and now 2.6.1 works perfect.

klaasnicolaas, 2023-08-12:
Why doesn't it work with the latest stable version of home assistant?

closed 2023-09-04 by github-actions[bot]