robbinjanssen/home-assistant-omnik-inverter #261: Hi i have a inverter Omnik 8000tl it's not on the list. did you have already the file to import value ?

Grenault76, 2023-12-06:
## Detailed description

## Context

## Diagnostics

```json
"Copy here the code or upload the file"
```

Grenault76, 2023-12-07:
> @Grenault76 <https://github.com/Grenault76> your question is a bit
> unclear. Unfortunately we don't know if the omnik 8000tl is supported, as
> no one has ever tried it before.
>
>    1. Please try installing
>    the component using HACS or manually.
>    2. Try the steps to determine your datasource
>    3. Please include the downloadable diagnostics from your inverter
>    device via the home assistant interface, so we can take a look

Hello, I just try the tree data sources possible

This that /js/status.js is the only method giving feedback see below
For the point 3 i have no idea where to find the downloadable diagnostic...

My best regards,
Gregory

192.168.1.17/js/status.js

var version= "H4.01.51MW.2.01W1.0.65(2018-02-271-D)";
var m2mMid = "639234596";
var wlanMac= "F0:FE:6B:C4:4E:20";
var m2mRssi= "100%";
var wanIp= "192.168.1.17";
var nmac= "F0FE6BC44E23";
var fephy= "off";
var yz_device_type="258";
var yz_device_num="1";
var init_inv_type="0";
var myDeviceArray=new Array();
myDeviceArray[0]="NLHN8020188M3006,V0.00Build,V6.01Build144,Omnik8000tl
,8000,0,779,304848,F00F01F02F03F04F05F06F07,1,";
;
function initPageText()
{
    var list = window.parent.reList("status");
    fileText("st1",list["t1"]);
    fileText("st2",list["t2"]);
    fileText("st3",list["t3"]);
    fileText("st4",list["t4"]);
    for(var i=1;i<=32;i++)
    {
        if(i!=14)
        {
            fileText("tx"+i,list[i]);
        }
    }
    init_main_page();
    initStatus();
    if(init_inv_type=="0")
    {
        hide("invTpDiv");
    }
    else
    {
        show("invTpDiv");
    }
    initInvList(parseInt(yz_device_type),yz_device_num);
    changeFont();
    child_getH();
}
function upfold(v)
{
    if(document.getElementById("up_"+v+"_div").style.display == "none")
    {
        show("up_"+v+"_div");
        document.getElementById("up_"+v).src = "image/up.png";
    }
    else
    {
        hide("up_"+v+"_div");
        document.getElementById("up_"+v).src = "image/down.png";
    }
    if(v == 3)
    {
        if(document.getElementById("ser_status").src=="")
        {
			//alert("mhtest");
            document.getElementById("ser_status").src="server_status.html";
[... 212 lines cut ...]
		    if(value*1==4294967295)
		    {
		        elem.innerHTML = "---";
		    }
		    else
		    {
		        sv= value*1/10;
			    elem.innerHTML= sv.toString()+" kWh";
		    }
		}
		else if (nv== 10)
		{
			if (document.getElementById("webdata_sn").innerHTML== "---")
				elem.innerHTML = "---";
			else
				elem.innerHTML = value+window.parent.reTip("5");
		}
		else
			elem.innerHTML = value;
	}
	else
	{
		alert(webdata_ids[nv]);
	}
}

robbinjanssen, 2023-12-11:
@Grenault76 great, this is the part we are interested in:
`myDeviceArray[0]="NLHN8020188M3006,V0.00Build,V6.01Build144,Omnik8000tl
,8000,0,779,304848,F00F01F02F03F04F05F06F07,1,";`

can you post this again, and then a screenshot with the actual values from the inverter?

robbinjanssen, 2023-12-11:
See https://github.com/robbinjanssen/home-assistant-omnik-inverter/issues/267

closed 2023-12-11 by robbinjanssen