robbinjanssen/home-assistant-omnik-inverter #273: buffer size too small (85 instead of at least 125 bytes)

Pakl8, 2023-12-30:
buffer size too small (85 instead of at least 125 bytes) using TCP

## Detailed description
During the setup of the Omnik inverter, I used TCP as data source, filled in the host IP address and serial number, an error appears, see snaphot, any suggeations?
[image]

MarijnS95, 2023-12-30:
I'm very sure we fixed this quite some time ago with https://github.com/klaasnicolaas/python-omnikinverter/pull/376

klaasnicolaas, 2023-12-31:
But we need to bump the package, to fix it in the integration.

Pakl8, 2024-01-01:
Thanks for your reaction, https://github.com/klaasnicolaas/python-omnikinverter is another package, I tried this one but had some issues during installation, this one was easier but run into a 3000TL inverter issue, which you explained.

klaasnicolaas, 2024-01-01:
You are not supposed to install the package, because it is not an integration. The integration uses this pypi package to retrieve the data, if we have performed a version bump then this issue will be resolved.

robbinjanssen, 2024-01-03:
Created a PR to bump the version: https://github.com/robbinjanssen/home-assistant-omnik-inverter/pull/276

robbinjanssen, 2024-01-03:
@Pakl8 please try the latest version :)

Pakl8, 2024-01-03:
For sure I must do something wrong, as I will get the same error, but I also do not see any changes in https://github.com/robbinjanssen/home-assistant-omnik-inverter/tree/main/custom_components/omnik_inverter exept for manifest.json, I used HACS by searching for Omnik Inverter Solar Sensor (No Cloud) to get things installed.

robbinjanssen, 2024-01-03:
@MarijnS95 @klaasnicolaas are we sure this has been released? https://github.com/klaasnicolaas/python-omnikinverter/releases

MarijnS95, 2024-01-03:
@robbinjanssen from the linked PR/commit there appears to be no tag pointing to it, so it wasn't released yet. Now `1.0.0` was published, and `1.0.0` shows up in the merge commit:

https://github.com/klaasnicolaas/python-omnikinverter/commit/b0498f228971c310d4a0cd5d626aa5a593280ea1

robbinjanssen, 2024-01-04:
Released https://github.com/robbinjanssen/home-assistant-omnik-inverter/releases/tag/v2.6.4

Pakl8, 2024-01-06:
Thanks for the update, works as it should be!

closed 2024-01-06 by klaasnicolaas